mavlink/qgroundcontrol #3335: Missing gps_helper.h header file

Darma, 2016-05-05:
Hi,

**Intro**
I am trying to build from master branch on (latest at this time) [`commit 79d97024ca219921c05962ee36cfca31c7c4d38`](https://github.com/mavlink/qgroundcontrol/tree/779d97024ca219921c05962ee36cfca31c7c4d38) Opened qgroundcontrol.pro via QtCreator, faced below Compile error while building it. 

**Compile Output:**

> 23:45:42: Running steps for project qgroundcontrol...
23:45:42: Configuration unchanged, skipping qmake step.
23:45:42: Starting: "/usr/bin/make" 
compiling ../qgroundcontrol/src/comm/LinkManager.cc
In file included from ../qgroundcontrol/src/GPS/GPSManager.h:26:0,
                 from ../qgroundcontrol/src/comm/LinkManager.cc:50:
../qgroundcontrol/src/GPS/GPSProvider.h:34:36: fatal error: Drivers/src/gps_helper.h: No such file or directory
 #include "Drivers/src/gps_helper.h"
                                    ^
compilation terminated.
make: *** [obj/LinkManager.o] Error 1
23:45:43: The process "/usr/bin/make" exited with code 2.
Error while building/deploying project qgroundcontrol (kit: Desktop Qt 5.6.0 GCC 64bit)
When executing step "Make"
23:45:43: Elapsed time: 00:01.

I have been compiled and running same master branch, [`commit ab89dc305c339335763f052803012488b664a069`](https://github.com/mavlink/qgroundcontrol/tree/ab89dc305c339335763f052803012488b664a069)  without any problem.

**Question?**
It seems `gps_helper.h` missing, or am I doing something wrong? Any help and suggestions will appreciated. Thanks.

---
Using: QtCreater (Community Version)
Based on Qt 5.6.0 (GCC 4.9.1 20140922 (Red Hat 4.9.1-10), 64 bit)

dogmaphobic, 2016-05-05:
You are missing a git submodule. Did you follow the build instructions?

```git clone --recursive https://github.com/mavlink/qgroundcontrol.git```

The `--recursive`` part collects not only the source tree but also all submodules.

Also, QGC does not yet support Qt 5.6, though this is completely unrelated to the problem above.

Darma, 2016-05-06:
Thanks, missing git submodule was the problem.

closed 2016-05-06 by Darma