mavlink/qgroundcontrol #3336: Home management: QGC sets home to Zurich even if neither GCS nor vehicle are there

LorenzMeier, 2016-05-05:
@DonLakeFlyer @dogmaphobic The home management seems out of whack, as it just defaults to a location which has nothing to do with either the GCS or the vehicle position. It should not show any home position if not configured to do.

DonLakeFlyer, 2016-05-05:
I assume you are not talking about vehicle home position indicator? You are talking about default map position?

DonLakeFlyer, 2016-05-05:
It will show the last home position it got from a vehicle if nothing else is available. Nothing means no vehicle and no GCS position. If that isn't available it shows default home position. Which at the moment I can't remember where it is.

Is it not showing a previous home position?

LorenzMeier, 2016-05-08:
There are two different discussion items:

   * Should a default (ghost) home position be visible at all? If yes, then it should be clearly marked as ghost position, e.g. by being transparent and having an uncertainty radius.
   * How should this position be initialised? For sure not by history, but if at all then by either the current GCS position (if on mobile) or just locked next to the first waypoint.

DonLakeFlyer, 2016-05-24:
Another possibility to add to the mix:
Allow the user to set a home position for the mission for real as an option. This would in turn create a SET_HOME_POSITION mission item in the mission. Very useful for fixed wing where arm position tends to not be where you want to come back to with an return to home.

kd0aij, 2016-05-24:
Please not that the "plan view" (A~B icon) inits with a window to set the "home position", but is apparently completely irrelevant to PX4. This was a major source of misdirection for me, as it lead to completely false expectations for the behavior of RTL.

LorenzMeier, 2016-05-24:
Its even the case for APM, where home is set on arming. So I just think that the whole dialog needs work to not imply something as configured which is in fact not.

LorenzMeier, 2016-05-24:
@DonLakeFlyer I think this shows that the attempt to allow consistent offline mission editing derailed the online experience quite a bit. We should really indicate what is going to happen, not what is convenient from the planning perspective. I'm happy to add the ability to programmatically set home for PX4, but I don't want it to lead to corner cases where QGC starts to always set it - it should be something (for PX4 and APM) that you consciously do.

kd0aij, 2016-05-24:
Is this related to the concept of "rally points"? Those I understand to be a set of alternative "home" points which are selected dynamically based on power and safety considerations.

LorenzMeier, 2016-05-24:
No its not. The main contributing factor is that Don tried to preserve the home position for offline planning, which doesn't really work yet.

DonLakeFlyer, 2016-06-14:
Made the wording better. Any larger change is too involved while we're trying to wrap up 3.0.

DonLakeFlyer, 2016-11-18:
This all needs to rethought through in 3.2

AndKe, 2016-11-20:
Some thought about home & adding autotakeoff:

1 User open QGC, locates the area where the planning is to be done (zooms in).
2 User adds WP - and home+takeoff is still at the old /previous place QGC have been used.

QGC newbie user would now see a line, zoom out , go and drag Home to there he actually started planning. (not even seeing that Takeoff item have been created out of sight) mission statistics are already displaying time/distance based on old home position.

an experienced user, would open mission item, "planned home position" and click "move home to map center" (or at least be aware of Takeoff and home influencing statistics)

I do not see an advantage of keeping old home position when planning new flight for nigher user..

This would be much more intuitive, and present the user with an easier to understand result:
1 User open QGC, locates the area where the planning is to be done (zooms in).
2 User create WP(or survey) - home/takeoff spawns in the middle of the visible map, (user sees it instantly, and are aware of both - moreover, mission statistics are instantly "correct" , or at least not off by very much)

So. please consider calling "move home to map center" when creating first WP/Survey item.

DonLakeFlyer, 2016-11-21:
Considering a revist of this for 3.1. @AndKe Thanks a ton for all your bug reports and change suggestions, very helpful.

DonLakeFlyer, 2017-03-25:
Completely redone: https://github.com/mavlink/qgroundcontrol/pull/4855

closed 2017-03-25 by DonLakeFlyer