mavlink/qgroundcontrol #3339: Map interaction painful / confusing

LorenzMeier, 2016-05-05:
This here is an user report, but my experience is similar:

For me on the MacBook it is almost impossible to use Mission Planing. Due to:
Still eratic two finger gesture if zooming, panning or trying a right mouse click!
I tried hard about 5mins to delete all my mission Items by selecting one-by-one and deleting them. I gave up after this worked out for about 2 out of 5. Please have a function which deletes a whole Mission.

dogmaphobic, 2016-05-05:
This is Don's domain, but as he's out reenacting *Deliverance*, here is one way to do it:

[image]

DonLakeFlyer, 2016-05-05:
Mac trackpad support is completely out of whack in Qt is you use map inside QQuickWidget. Issue already in on this.

DonLakeFlyer, 2016-05-05:
I had remove all on the hamburger menu as well before. I moved it to sync. Maybe having it broth places would be better.

LorenzMeier, 2016-05-08:
Yes, both would definitely help. People don't expect it under sync.

DonLakeFlyer, 2016-05-30:
I turned off two finger pinch on Mac due to Qt bug. Two finger drag scroll still works to zoom in/out.

closed 2016-05-30 by DonLakeFlyer