mavlink/qgroundcontrol #3342: Preferred link handling bug

LorenzMeier, 2016-05-06:
If QGC looses the preferred link it will continue to try to use it, even if the preferred link is disconnected and it has still an alternate link.

To repro:

- Connect QGC via radio and USB
- Remove USB, power autopilot via battery
- Try to set params. Every set attempt will fail

dogmaphobic, 2016-05-06:
This is a function of the link "connected state". The link is never (automatically) disconnected so it stays as the preferred link. Were it to disconnect, the preferred state would go with it and the alternate link would take over.

DonLakeFlyer, 2016-05-07:
Auto-disconnect for usb pull should be working. Link should go away.

DonLakeFlyer, 2016-05-29:
This is a problem with parameter communication only. I need to think about this some more. I'm not quite sure yet how to make the parameter protocol work with links getting ripped out from under it.

DonLakeFlyer, 2016-05-29:
@LorenzMeier Running this by you for a double check since this multi-link stuff can get very confusing: Things like the Mission read/write protocol are essentially stateful hence they need to complete an invididual protocol on the same link. Each individual read/write protocol sequence picks up the new priority link at the start of the sequence. So MissionManager has the concept of a dedicated link which is set/associated with each mini-protocol sequence.

Wheras the parameter protocol is essentially stateless as far as a ground station is concerned. It can send REQUEST_LIST on one link, and then say ask for missed params on initial load on another link without causing any problems. Current ParameterLoader has a dedicated link concept like MissionManager which is incorrect since it is never updated. But I don't think it is needed at all.

LorenzMeier, 2016-05-29:
Removing the link concept could be sufficient to fix the usage problems.

DonLakeFlyer, 2016-05-29:
The preferred link handling from a vehicle standpoint is working fine. It switches when the link goes away. The problem is you can't just always use the preferred link for all communication. Mission communication already uses the new preferred link each time is starts a read/write sequence. So that will switch correctly when the link changes. So for that case when say one write fails, the next will succeed since it will switch links. You just can switch links mid-stream.

closed 2016-05-29 by DonLakeFlyer