mavlink/qgroundcontrol #3345: MAVLink param names can be 16 chars wide in editor, but get cropped

LorenzMeier, 2016-05-08:
To repro run the PX4 fixed wing config (SYS_AUTOSTART set to 3033). These params are 16 chars long and get cropped on a 13" screen:

[image]

DonLakeFlyer, 2016-05-08:
This doesn't look to be from a latest build. It doesn't have the search thing on the top. Maybe it's missing some of Gus' changes? The column width is set to 20 char widths which would seem to work fine. But I can't test on Android.

DonLakeFlyer, 2016-05-12:
@LorenzMeier Can you retry this with latest builds?

DonLakeFlyer, 2016-05-23:
@LorenzMeier Retest?

DonLakeFlyer, 2016-05-30:
I tried latest build on my 13" and my 9". Fine on both.

closed 2016-05-30 by DonLakeFlyer