mavlink/qgroundcontrol #3348: Users confuse autopilot and mag orientation setting in mag dialog

LorenzMeier, 2016-05-08:
We need icons or some other visually clear differentiation between autopilot and mag. Users keep confusing the two, in particular since the mag calibration asks in the first step for the *AUTOPILOT* orientation, but the user's mental model is all about the mag (and so in user tests a lot of them select the mag's orientation).

[image]

DonLakeFlyer, 2016-05-23:
The ArduPilot screen looks like this
[image]

Is that better?

LorenzMeier, 2016-05-23:
Much better. Way too much text on the PX4 side and the description is not really user-parseable.

LorenzMeier, 2016-05-23:
Also the mag cal pops a right dialog for the autopilot orientation. I would not do that. Better approach would be to pop the autopilot orientation dialog for ACCEL cal only and to condition the mag cal button click ability on accel being finished.

closed 2016-05-29 by DonLakeFlyer