mavlink/qgroundcontrol #3351: If tile loading fails once, it doesn't necessarily recover

LorenzMeier, 2016-05-08:
@dogmaphobic See the screenshot I sent via email. I have this on multiple devices.

dogmaphobic, 2016-05-09:
Failed tiles currently must be manually "retried". In the majority of the cases during my tests, the problems were related to tiles simply not existing at all, which would make automatic retries futile. I will see what can be done to differentiate from a 404 to a lack of network connectivity.

LorenzMeier, 2016-05-09:
I did zoom in an out and it would never resolve the missing tiles on this tile level. If you ever boot QGC with internet off at a specific location you can "infect" that specific zoom level by never getting those tiles ever again, unless you completely wipe out the QGC settings.

dogmaphobic, 2016-06-07:
I think this was related to the fixes in #3492. It was not related to offline map caching. Rather, it was likely using old URLs for fetching those tiles. I just tried it and once I "connected to the Internet", hitting "Resume Download" proceeded to download all tiles with no errors.

closed 2016-06-07 by dogmaphobic