mavlink/qgroundcontrol #3356: Log download 10 times slower then downloading onboard file

sanderux, 2016-05-09:
When downloading a log through the "log download" interface a 1MB log takes about 1.5 mins.
The same file downloaded though "onboard files" takes only 10 secs or so.

sanderux, 2016-07-18:
@DonLakeFlyer @dogmaphobic @LorenzMeier any update on this?

DonLakeFlyer, 2016-07-20:
I'm not sure log download uses "burst" style packet sending.

sanderux, 2016-07-23:
@DonLakeFlyer could you not use the same method in log download like on onboard file download?

dogmaphobic, 2016-07-23:
We had to implement the log download to match the way APM does it (for it to be compatible). All other things being equal, the main difference is the payload size. Onboard Files (FTP) has a payload of 251 bytes while Log Download has a payload of only 90 bytes. That is, a packet for each 90 bytes of log data.

dogmaphobic, 2016-07-23:
> I'm not sure log download uses "burst" style packet sending.

Yes, it does in the same way FTP does it. It also keeps track of "chunk integrity" so the client side can recover from missing chunks (data errors).

NaterGator, 2016-07-23:
I adjusted the mavlink payload size for log download and it sped things up significantly. I mentioned this to @LorenzMeier on gitter but I don't think he was interested in making the change unless the performance boost could be explained. The mavlink module onmodulis designed to optimally fill the transfer buffers with burst, but clearly there is some protocol or OS overhead we are not correctly  accounting for.

dogmaphobic, 2016-07-24:
> I adjusted the mavlink payload size for log download and it sped things up significantly. I mentioned this to @LorenzMeier on gitter but I don't think he was interested in making the change unless the performance boost could be explained.

I experimented with the code on the firmware side. I set the streaming burst limit to 1M Bytes to see how it would handle it. The code just set on the *send* loop, sending packets non stop (burst mode) and it made no difference. That would suggest the bottleneck is the number of (tiny, 90-byte) packets sent.

NaterGator, 2016-07-24:
I'm also implicating either the NuttX or Windows CDC-ACM driver. For whatever reason there seems to be a decent overhead per-packet or per-transaction that makes the 90 byte buffer particularly expensive. Prior work by others suggests we should be able to achieve greater than about 500KB/s over USB CDC... and we are nowhere near that.

sanderux, 2016-07-24:
I noticed this on ubuntu 14.04
So it's not windows specific

dogmaphobic, 2016-07-24:
Mac OS here. Certainly not OS specific.

kd0aij, 2016-08-17:
1. release 1.4.1 firmware: new logger files (extension .ulog) don't appear in the log download window

2. The onboard files widget reports:
[image]

NaterGator, 2016-08-17:
The slowness should have been fixed by https://github.com/PX4/NuttX/pull/74/commits/cc1fe911ce453d6e463109afef3eacd199dd98a3, https://github.com/PX4/Firmware/pull/5222/commits/547592fddf75d0d7a97fa983e55ef1889257e635, and https://github.com/mavlink/qgroundcontrol/pull/3929/commits/b4afd5283bff3291f1bfabfc06d99e8af9c63e84.

@kd0aij can you open a separate issue?

DonLakeFlyer, 2016-08-18:
@NaterGator So, close this one now?

NaterGator, 2016-08-18:
I think so. I haven't heard reports of issues with those changes. I think there is still room for improvement in the NuttX FAT driver / firmware data management, but things are much better now.

closed 2016-08-18 by NaterGator