mavlink/qgroundcontrol #3359: Just clicking the arming slider without moving it executes it already

LorenzMeier, 2016-05-09:
@DonLakeFlyer Pretty dangerous: If you just click the slider without moving it you get the same result as if you move it all the way to the right.

shadow1runner, 2016-05-10:
Honestly, I always saw this as a feature - cause dragging the slider when you only have a notebook is really tedious and it's not a great UX either -- I preferred the yes/no buttons here by far.
Of course I agree that on a touch-enabled device the slider should work as expected and honestly, a slider is not a control meant for non-touch enabled devices in general.

LorenzMeier, 2016-05-10:
@shadow1runner I understand where you are coming from, but a clickable slider is not the way how you do this properly - it would mean that we offer a completely different UI element for desktop, which in particular ensures that the click location is at a distance from the first button to ensure no fat-finger confirmations occur.

I'm closing this for now - I use the slider on a laptop with the trackpad just fine. However, feel free to file an issue for a desktop-specific confirmation method to be added.

DonLakeFlyer, 2016-05-12:
This is busted. You are supposed to need to slide all the time no matter what the OS/Device.

DonLakeFlyer, 2016-05-13:
@LorenzMeier Not sure how you are making this happen. Just clicking it dismisses the command for me. I can't make it confirm the command.

LorenzMeier, 2016-05-13:
The reason it was closed is that I already fixed it on master.

closed 2016-05-13 by LorenzMeier

DonLakeFlyer, 2016-05-13:
Got it. Can you make sure to reference across to the issue on the pull so everything gets connect together.