mavlink/qgroundcontrol #3362: Guided Bar: Arm Button not visible on initial connect

shadow1runner, 2016-05-10:
I've only tried it on the APM stack, so I guess that this works on PX4, but on APM the home position is set on initial arm.
The code lines were last changed by @LorenzMeier in 4cfd5a1d:

```
QGCButton {
    text:       (_activeVehicle && _activeVehicle.armed) ? (_activeVehicle.flying ? qsTr("Emergency Stop") : qsTr("Disarm")) :  qsTr("Arm")
    **visible: (_activeVehicle && _activeVehicle.homePositionAvailable)**
    onClicked:  _guidedModeBar.confirmAction(_activeVehicle.armed ? (_activeVehicle.flying ? _guidedModeBar.confirmEmergencyStop : _guidedModeBar.confirmDisarm) : _guidedModeBar.confirmArm)
}
```

shadow1runner, 2016-05-10:
I've tried fixing it, but `_activeVehicle.autopilot` seems to be empty, at least on ArduCopter (SITL), can't try PX4, unfortunately

LorenzMeier, 2016-05-10:
@shadow1runner You can try PX4 fairly quickly by running SITL, which is supported on Linux and Mac OS: http://dev.px4.io/simulation-sitl.html

The intent of my change was to deny arming before the vehicle has position lock (because arming via tablet without a valid position source is just dangerous). We would need to expose that property from the vehicle.

LorenzMeier, 2016-05-10:
Fixed in 24c4e9d.

closed 2016-05-10 by LorenzMeier