mavlink/qgroundcontrol #3363: RTSP/HTTP video stream to QGC on Windows

Kiwa21, 2016-05-11:
Hello all,
I am currently trying to use a regular IP camera to feed a video stream into QGC (Windows) but with no success until now. I have seen that the actual Video plugin uses some kind of gstreamer dependencies. Unfortunately, I don't know how/where it searches these since there are no pkg finder on windows.
I have access to the video stream via RTSP or HTTP (I can verify it form internet browser or VLC).
The video is already encoded on a h264 format.

Is it possible to feed this kind of video directly into QGC widget ?
Do I need a MAV to be connected to enable the video widget or can it run even without any MAV connected ?

Wouldn't it be easier to let the user select between UDP or input a RTSP/HTTP url ?

Thanks in advance for your help/suggestions,
BR

DonLakeFlyer, 2016-05-12:
Description of video support is here: https://github.com/mavlink/qgroundcontrol/blob/master/src/VideoStreaming/README.md <https://github.com/mavlink/qgroundcontrol/blob/master/src/VideoStreaming/README.md>

uavcam, 2016-08-04:
HTTP stream is nonsense
RTSP stream works in theory - i wrote the code already. but no time to build controls etc. to make it a clean solution for qgc in addition to UDP

booo, 2022-11-01:
RTSP streams should work on windows now.

closed 2022-11-01 by booo