mavlink/qgroundcontrol #3364: Log download crashes with many entries

LorenzMeier, 2016-05-12:
If you have 120+ logs QGC will crash on trying to display them:
http://discuss.px4.io/uploads/default/original/1X/875efb3202959946ef26b7e1ab28795e16207e5e.pdf

andrejpan, 2016-05-18:
It is also happening when there is less logs. I had around 90 logs and I used qgc 2.9.7 at ubuntu 14.04. Here is the print screen: [image]

DonLakeFlyer, 2016-05-23:
This is the Log Download widget crashing?

andrejpan, 2016-05-24:
Yeas, but it is also killing QGC 297b.

dogmaphobic, 2016-06-07:
I was able to reproduce this and I *think* I know where the crash is. The (bulk) change from model to view is running into some race condition in the QtQuickWidget boundary. Given that it is quite unlikely Qt will ever fix these QtQuickWidgets issues, I have to come up with an alternate way of handling this.

DonLakeFlyer, 2016-07-01:
@dogmaphobic Any update on this one?

dogmaphobic, 2016-07-01:
Nope. Until we switch away from QtQuickWidget, I suggest not accumulating an inordinate amount of log files. In the mean time, it is best to take the SD card out and load it directly onto the computer. Log transfer through MavLink sucks any way due to the tiny packet size.

DonLakeFlyer, 2016-07-08:
@dogmaphobic Is there a reason this doesn't just use QmlObjectListModel?

DonLakeFlyer, 2016-07-08:
There's more than just a refresh crash here. I just crashed booting QGC with the log download widget active. It crashed trying to clear the list on the set active vehicle.

DonLakeFlyer, 2016-07-08:
Of course it was a one time thing.

DonLakeFlyer, 2016-07-08:
It's not crashing in the list clear. It's crashing in the disconnect from the previous vehicle. Something is out of whack since there is no previous vehicle.

DonLakeFlyer, 2016-07-08:
@dogmaphobic The repro is connect/disconnect/connect with widget up. It's crashing because setActiveVehicle is not handling setActiveVehicle(NULL) corrrectly when a vehicle goes away. It still keeps old pointers from a deleted vehicle. The second connect then tries to disconnect signals on a bad pointer.

NaterGator, 2016-07-08:
The design pattern around activeVehicleChanged explicitly disconnecting each connection by matching signal/slot has always bothered me. 

Why not store a QList<QMetaObject::Connection> of each connection made when a new vehicle is connected and simple iterate that list when disconnecting the old one?

DonLakeFlyer, 2016-07-08:
Not quite sure what you are talking about? What connections are you talking about? The C++ connects to all sort of random stuff on a Vehicle. In this case the connection is actually to UASInterface.

DonLakeFlyer, 2016-07-08:
@dogmaphobic I can repro some variants of the large list crash. My guess on what is happening is two things:
1) You aren't marking C++ object ownership on the entries in your ListModel. Like this: https://github.com/mavlink/qgroundcontrol/blob/master/src/QmlControls/QmlObjectListModel.cc#L162
2) Also you are calling delete instead of deleteLater on the entries you are removing from the list model. This tends to confuse Qml since it has a delayed dereference/cleanup way of doing things.

I'm not entirely convinced it has anything to do with QQuickWidget. Do you have direct evidence of that? I ran into these types or problems with my initial implementation of QmlObjectListModel. My suggestion is to use QmlObjectListModel since it does the correct Qml/C++ object ownership marking. And to use QmlObjectListModel::clearAndDeleteContents to delete the list since it does the right thing with calling deleteLater on the inner objects. It might fix the issue.

To me this is a pretty critical bug to fix since you never need logs until you really need them. So it's very likely there will be whole big pile of them there from all the flights with non problems. There are also many cases where you can't pull the sd card and get the log manually since it is deep in a Vehicle and you can't get to it. I myself have a fixed wing like that.

DonLakeFlyer, 2016-07-08:
I add the ownership and deleteLater to the list model code in the controller and I no longer can repro a crash on a 122 count log list. I think that's the safest for 3.0. For 3.1 we should switch to QmlObjectListModel so we pick up any other crazy Qt wierdness that gets implemented in there.

dogmaphobic, 2016-07-08:
Cool! I get here and find that the issue was found and fixed before I even get my coffee 8) Thanks!

The reason I thought it had something to do with QQuickWidget was because the crashes were within that module when looking at the stack trace.

> Is there a reason this doesn't just use QmlObjectListModel?

There was one but I can't remember. There is some discussion in #2518 but that was much later than the Log Download code.

DonLakeFlyer, 2016-07-08:
I would have had no idea if it didn't smell like problems I had already run into. Can you fix the setActiveVehicle crash?

closed 2016-07-09 by DonLakeFlyer