mavlink/qgroundcontrol #3367: Flight view UI performance issues on standard hardware

LorenzMeier, 2016-05-13:
@DonLakeFlyer @dogmaphobic The flight view is only smooth on my i7 quad core 17" MBP. My 13" dual core MBP lags (e.g. on the arming slider). Both devices are just over a year old, so pretty up to date.

I think we should try to figure out where the performance bottleneck is coming from - if we can't find anything quickly then maybe rather for the 3.1 release.

NaterGator, 2016-05-13:
Are you specifically referencing dragging performance or do you mean rending in general is sluggish?

DonLakeFlyer, 2016-05-23:
@LorenzMeier Fly is is not smooth when doing what?

LorenzMeier, 2016-05-23:
I think I hunted it down by now to GPU load. I'll close this. It works when I'm not running the rendering part of SITL.

closed 2016-05-23 by LorenzMeier