mavlink/qgroundcontrol #3372: Summary UI

cfelipesouza, 2016-05-13:
Continuing the UI Layout Reorg done by @dogmaphobic on #3333 and due to Usability Tests, we are proposing some improvements on Summary Screen.

We've done a basic facelift on this screen, that we would like to continue through others. Our proposal is based on legibility and visibility, giving to the user a more visible information (using contrast, hierarchy, bigger elements, etc,), but still maintaining the current visual identity.

Also, we created a [new icon](https://github.com/Dronecode/UX-Design/tree/gh-pages/QGC/Application/new_icon_set/summary) set that will guide, in the future, the others icons. It is based on the new QGC Branding, using lines in its drawing. 

We are sending the screens with some specifications to help on implementation. Any doubts or comments, please, reply me. We hope the way was done might be useful.

**Summary - Home**
[image]
[image]

**Summary - Hovers**
[image]
[image]

**Summary - Warnings**
[image]
[image]

@anselmolsm @biapalmeiro @ricardogejao

dogmaphobic, 2016-05-14:
This is perfect. Before going into it (and requesting the artwork so it can be implemented), have you given any thought to the *Light Palette*? Under *General Preferences*, select the *Outdoor* Style.

There is also a Widget Don put together to look/edit the palette. It's a good place to take a look at the current set so you will have a better idea how to match what you see to the palette being used.

[image]

cfelipesouza, 2016-05-16:
The ideia proposed is more based on layout improvements (structure and info's reframe) than colours, despite knowing that we are changing a little bit some of them. So, for now, you can only change Dark Palette and maintain the same Light Palette while we go deep on it. What do you think?

About the widget, I'll take a look to match the current set.

The icons are [here](https://github.com/Dronecode/UX-Design/tree/gh-pages/QGC/Application/new_icon_set/summary), artwork is [here](https://github.com/Dronecode/UX-Design/tree/gh-pages/QGC/Application/screens/summary) and specs were posted above.

cfelipesouza, 2016-05-19:
Ok, so here I am again :)

Thinking about Light Palette, I changed some colours to be possible to use them in both themes. I put away the yellow and brought the same blue of QGC Branding, for example.

I'm pasting here the idea for Light theme, but I would like to hear from you what do you think about it, before post the specs and update the artworks.

[image]

dogmaphobic, 2016-05-19:
The intent of the light palette is for using outdoors. It needs a lot of contrast to compete with the intense ambient light. I cannot test it right now (not only I am in the middle of a conference all but also, it's quite dark outside) but you can do that yourself. Just load this page on a tablet, bring up the screen grab above and bring the tablet outside. You should be able to read it on a sunny day.

I will work on this (the original, dark palette) this weekend. By then I will have a better idea about how the light palette works.

With that said, I agree with the changes. I just feel there is a need for greater contrast between text/graphic elements and background.

cfelipesouza, 2016-05-19:
In fact, I already went outside with a tablet and a computer to see how the palette works. This proposal is fine for me, but it will be good to have your opinion on it as well, because you have a better idea about everything and already have passed through other issues.

If you realize that we need more contrast, I will be glad to work more on it. :)

DonLakeFlyer, 2016-05-19:
I'm not so hot on the on the lighter color for the Labels in "Label: Value" pair visuals. It seems hard to read. It's going be be especially a problem on the outdoor palette. Also we have a disabled palette color for labels which if I remember correctly is used a couple places in setup screens to show disabled elements. This color is very close to this color. Another comment would be that any "Label: Value" pair style change needs to be carried across the entire UI not just this page. This paradigm is used in a number of other places such as: toolbar dropdowns, parameter editor side dialog, others.

Also keep in mind that any palette changes to make this screen look right like for example, hover/select colors are global to the app. So the full app/palette needs to be tested against the other ui as well. I'm perfectly fine with changing things, just need to make sure it works everywhere.

Another comment is the Red for the incomplete setup warning. I wonder if that would be the same color as the yellow/orange error message popup color to keep things inconsistent.

DonLakeFlyer, 2016-05-19:
A possibility for the "Label: Value" pair visual is to maybe instead of using a color difference for the label. Make the Value so semiBold font to make it stand out a little more.

Another thing to think about: Original the buttons had circular red/green indicators. In order to save space, I got rid of those and just changed to icon to Red on the button. Now that the little summary panels are using the icon as well, may want to think about just coloring the icon there as well and toss the circular indicator.

cfelipesouza, 2016-05-23:
Thanks for your comments @DonLakeFlyer .
- We are going to test about Semibold for values. It could be interesting, in fact.
- Our idea for this moment is to maintain the same appearance of QGC. Before deleting the circular indicator, we would like to propose and test other things. But this will be round 2.
- I know that changing the palette will affect the entire app. That's why we're proposing little changes instead of changing everything. But you can help us saying where wouldn't work, for example.

closed 2024-06-02 by HTRamsey