mavlink/qgroundcontrol #3376: The log size shows is incorrected

glody, 2016-05-16:
Test Environment
QGroundControl: Daily Build: v2.9.7b-1-g1fb0f04
Windows: Windows 8.1 64 Bit Laptop
Linux: Ubuntu 14.04 64 Bit Laptop
OS X: OS X 10.11.2 MacBook Pro
Purpose:
Check if the logs is able to download normally
Step:
1. Click Widgets and select Log Download
2. Click Refresh on log window(If here still hasn't logs list, please do some operation in QGC then open this window)
3. Select a log and download it.
Excepted result:
The log size shows in 'Status' can not bigger than shows in 'Size'
Actual result:
The log size shows in 'Status' can bigger than shows in 'Size'
[image]

dogmaphobic, 2016-05-26:
I cannot reproduce this. Not even sure how it could happen. The size reported is the number used for the requests. It would never request more than reported. Even if the log is growing during the download. Here is a screen grab showing right at the end of a download, before it completed. Once complete, the Status changes to *Downloaded*.

[image]

NaterGator, 2016-05-26:
Just thinking out loud... if multiple links are present I could see this being possible. Also, we don't check if a chunk has already been written before writing again, and when we write again we increment the byte count by the chunk size (https://github.com/mavlink/qgroundcontrol/blob/4af575609f92be2c4ca7edf19ff4212c1998928a/src/ViewWidgets/LogDownloadController.cc#L349) so that would be the obvious source of this issue. 

The simple solution is to drop chunks that have already come in.

glody, 2016-05-30:
Checked on v2.9.7b-46-g4af5756, this issue unable to reproduce. close this. thanks.

closed 2016-05-30 by glody