mavlink/qgroundcontrol #3378: Virtual Joystick can't control PX4 flight stack.

SamChenzx, 2016-05-16:
I'm using Qgroundcontrol 2.9.7b as my ground station and I want to use the virtual joystick without a real RC or joystick radio by setting COM_RC_IN_MODE to 0(Joystick/No RC Checks). I can arm motors by command line. The problem is when I attempt to takeoff my quadrotor in manual flight mode which runs PX4 flight code, no matter how I move the virtual joystick, the PWM output seems always be 0, I want to know that do I need extra setup before using virtual joystick, or virtual joystick can not control a real quadrotor?

dagar, 2016-05-16:
Try COM_RC_IN_MODE 1

SamChenzx, 2016-05-16:
When I change COM_RC_IN_MODE to 1, as I don't have a real RC radio
connected to quadrotor, QGC warns that:
Detected 0 radio channels. To operate PX4, you need at least 5channels.
And the Radio menu turn into red color. Does it works?

andrejpan, 2016-05-18:
Hey,

I set up pixracer and QGC 2.9.7 with psp3 gamepad. COM_RC_IN_MODE has settings Joystick/No RC Checks (I do not see any number with this parameter any more). There is also settings for a joystick, QGC should recognize type of gamepad and calibration should be also made. 
[image]

And I think this is it. In few days I will also try QGC 2.9.7beta and I can report if gamepad control still works.

LorenzMeier, 2016-05-18:
@SamChenzx You need to arm via QGC to control the system. You cannot arm it with the virtual joysticks. It sounds like you're trying to do that.

andrejpan, 2016-05-18:
Currently I am arming/disarming motors with virtual joystick with no problem, should we make another issue?

SamChenzx, 2016-05-18:
@LorenzMeier Yes, that's the problem, I can't arm it with the virtual joysticks. Besides, seems can't arm it via QGC, as in the screen shot, QGC just start with a white circle with no arm/disarm option.
[image]

LorenzMeier, 2016-05-18:
QGC does only allow you to arm if you have GPS. The reason is that flying the system reliably without any position input is very hard for untrained pilots. We only recommend to fly the system in POSCTL mode with thumb sticks.

SamChenzx, 2016-05-18:
With a GPS lock, it't much reliable to control vehicle. But how about the indoor application, we are not so easy to acquire GPS information in a hermetic room.

SamChenzx, 2016-05-19:
Other issues here are as follows:
Firstly, QGC sometimes can't arm quadrotor even with GPS locked, as using virtual joystick, QGC warns that:Arming DENIED. Manual throttle non-zero. At this situation, my virtual joystick's two thumb pads are both at zero point actually. Does it mean I should pull the left thumb pad down to make throttle to zero?
[image]
Secondly, when I arm it via command line, the output PWM signals seems always have one motor or two at a high duty cycle, but which motor can be high duty varies from time to time(shows in the below screen shots, can be motor 2 or sometimes motor 4), though I still leave thumb pads at zero point and keep my quadrotor horizontal and stable on the ground. I have no idea it's an issue of QGC or PX4 flight code.
[image]

[image]

closed 2016-06-14 by DonLakeFlyer

MathiasGilson, 2016-08-13:
Found it ! You need to disable the safety switch from the parameters: put the param **CBRK_IO_SAFETY from 0 to 22027**. This will simulate that the safety switch is always on and you will be able to arm it ;)

DonLakeFlyer, 2016-08-13:
@LorenzMeier Shouldn't firmware send better error to ground station about failure to arm if it's the safety switch. Otherwise user has no idea what the problem is?

dagar, 2016-08-16:
I still find arming and mode change rejection errors bewildering.
https://github.com/PX4/Firmware/issues/4663

sneiman, 2016-10-18:
I have this problem. cf2 running px4. QGC 3.0.1 on ubuntu 16.04. flight mode set to stabilized, CBRK_IO_SAFETY = 22027. using taranis 9d plus as joystick. cfbridge to QGC to MAVLink to Crazyradio PA. Can only arm using flight screen ARM button. Can fly once I do this. Its a problem for my specific development - working on swarms of cf2s and a custom 130 for both controlled and preplanned follow flight - REALLY need to make things happen from controller. Am I missing something? thanks

DonLakeFlyer, 2016-10-19:
Do you get no response from the Vehicle. Or does it send back that it rejected the arm?

sneiman, 2016-10-19:
I get nothing - no messages on flight screen,  or in console, or in status bar. Am I correct in trying to arm with throttle at 0, and yaw/rudder extreme right for 5 seconds?

On a minor note, when using a joystick with 0 throttle at extreme bottom, system calibrates properly but requires the throttle stick to be moved each time QGC is started to show correct stick position/value. Just fyi.

DonLakeFlyer, 2016-10-20:
What version of QGC are you using?

sneiman, 2016-10-25:
3.0.1 on Ubuntu 16.04.

This entire message is confidential. If it isn't intended for you, you may not use it – so please throw it away and forget about it.

metinburak, 2016-11-22:
Hi, when we arm and takeoff via QGC, then use virtual joystick in POSCTRL mode, QGC sends the manual control mavlink messages to Pixhawk through the telemetry 1. When we use the telemetry 2 port of Pixhawk (in POSCTRL mode) with a companion computer on the drone and send the same mavlink messages to Pixhawk as what the virtual joystick sends, we can not control the drone. Does the POSCTRL mode only accept the manual control messages from Telemetry 1?

SamChenzx, 2016-11-22:
Hi, Burak

If use companion computer, you can try offboard mode to control the drone.

DonLakeFlyer, 2016-11-22:
> QGC sends the manual control mavlink messages to Pixhawk through the telemetry 1. When we use the telemetry 2 port of Pixhawk (in POSCTRL mode) with a companion computer on the drone and send the same mavlink messages to Pixhawk as what the virtual joystick sends, we can not control the drone.

@metinburak You are going to have to ask the firmware folks about that. Not really a QGC question.

Mocart96, 2018-05-01:
hello @metinburak
I know topic is old but did you find an answer on Telemetry2 port issue? 
Could you explain me little more what means "we can not control the drone"?

sneiman, 2018-05-01:
sorry - not much help from me. i stopped using this setup some time ago before fully resolving this issue. good luck

Sent from mobile device - sorry for typos and brevity.

This entire message is confidential. If it isn't intended for you, you may not use it - so please throw it away and forget about it.

apurvadey, 2020-01-06:
Hey have you been able to to use the virtual joystick without a real RC or joystick radio by using COM_RC_IN_MODE ??