mavlink/qgroundcontrol #3381: Connection loss info can create false positives

LorenzMeier, 2016-05-18:
@DonLakeFlyer I saw this on USB as well. This video of a user shows a situation where he has perfect control over the drone via USB Gamepad while QGC says its lost its connection:
https://www.dropbox.com/s/m92x7edia447grz/2016-05-18%2017.39.05.mp4?dl=0

DonLakeFlyer, 2016-05-30:
I did a code review of this and don't see any holes. If QGC stops seeing heartbeats from the vehicle it will show communication lost. That doesn't necessarily mean that commands to the vehicle will stop working.  This link is still open and QGC is still active. It just means something strange has happened on the vehicle side such that heartbeats have stopped.

The key to determining whether there is something wrong with QGC heartbeat processing is to check mavlink inspector for heartbeats when you are in this state. But without a repro I can't do much here.

DonLakeFlyer, 2016-06-14:
Unless I can get a repo that shows that QGC is incorrectly reporting communication lost I'm assuming the vehicle stopped sending heartbeats for some reason.

closed 2016-06-14 by DonLakeFlyer