mavlink/qgroundcontrol #3384: Suggest QGC to support FPV video stream, add a configure menu to set rtsp address

SamChenzx, 2016-05-21:
QGC is a powerful GCS to control PX4 flight as well as get real time video stream. I suggest to add a configure menu so that we can connect to tsp stream over wifi, like rtsp://192.168.1.1/fpvview, especially for snapdragon flight, like the follow picture.
[image]

DonLakeFlyer, 2016-05-23:
@dogmaphobic This is all greek to me...

dogmaphobic, 2016-05-23:
Technically we can support anything. It’s all software and the core video handler is in place. I built a hardwired pipeline for simplicity’s sake. One that satisfied my own setup. Adding the ability to create different video pipelines, so it can handle things like RTSP would just a matter of adding it to the code. The trick is that building these pipelines are quite complex. With the low latency requirements, it’s crazy making complex. If there is a way to make it simple for users, I can add it to the UI. Also, the current video streaming, which uses GStreamer at its core, only includes the parts of GStreamer that are needed. If we allow arbitrary pipelines, we will need to include all of GStreamer. Not an issue for desktop but a bit heavy for mobile.

This has been in my back burner since before I started. I just needed a use case that would justify the amount of work involved. I also don’t have anything that transmits video in a different way than what I implemented. I messed around with RTPS on the sender side but the latency was too great to be usable. That was an issue within the Odroid SBC I used more than anything I guess. Either way, tapping straight into the raw h.264 stream coming off the camera was the best (latency wise) way I found.

SamChenzx, 2016-05-23:
Thanks for your reply, Gus! As far as I know, in many applications, latency
up to 1 second or more still acceptable. In mobile side, the GPU can handle
several types of stream. So what if we just add two or three most popular video
stream type for users?

booo, 2022-11-01:
QGC has a menu to configure RTSP streams now.

closed 2022-11-01 by booo