mavlink/qgroundcontrol #3385: Suggest QGC's virtual joystick, especially the throttle stick can keep still when lose hold of it.

SamChenzx, 2016-05-21:
I find it's quite difficult to control PX4 flight use virtual joysticks, especially on Linux side, users can't pull throttle down and side to arm at the same time, on android devices this can be done due to multi touch support. I find a simple GSC shown in the below, which is much easier for users to control PX4 flight via virtual joystick as it will keep still when I lose my hand on throttle stick. The cross-stick is much simple and easy to operate. I think it will be much better if the virtual joystick is easier to operate.
[image]
[image]

LorenzMeier, 2016-05-23:
I don't understand the core issue - what part is not responsive?

SamChenzx, 2016-05-23:
@LorenzMeier Sorry for the unclear description. The core issue is when I start up QGC, both left and right virtual joysticks will keep at the center, and if I let go of the joysticks, they will return to center. It's good for the right virtual joystick.
However, if the left joystick's vertical direction is used for controlling throttle, I think it will be better to start at the lowest position, shown in the below picture:

[image]

Besides, when let of it don't automatically return to center but keep the vertical position where it is, as the referred GCS or a real joystick do. If both virtual joysticks start up at center,  users who use virtual joysticks can't arm vehicles by  sliding arm button in QGC, as the initial throttle not at the zero point. Please check it, thanks!

DonLakeFlyer, 2016-05-23:
Virtual joysticks are mainly meant to be use with a non-manual flight mode such as Position Control. In this case centering makes more sense. I would suggest arming using the Arm button instead of the virtual sticks.

SamChenzx, 2016-05-26:
@DonLakeFlyer  Those days we have tested snapdragon flight runs PX4 outdoor and with GSP locked. The issue is that I can switch between Manual, Acro, Stabilized, Rattitude and Altitude Control, while Position Control and other flight mode can't be used, don't know why but QGC warns "Set mode command temporarily rejected".

DonLakeFlyer, 2016-05-26:
The comes from the Vehicle so you need to ask on the Firmware side

closed 2016-05-27 by SamChenzx