mavlink/qgroundcontrol #3391: Arm button should not be hidden if no GPS

DonLakeFlyer, 2016-05-23:
This isn't something that QGC should be handling. For example on ArduPilot you can turn off no-GPS arm failure with a circuit breaker. If I have that turned off I should be able to arm from the button. If firmware wants to fail arm with no GPS it should fail the arm and return a user friendly error. The current QGC implementation is very firmware specific to a specific stack/vehicle type. Another example is that ArduSub has no GPS I would guess. Now you can't arm that either.

DonLakeFlyer, 2016-05-23:
Other option is mavlink generic mechanism to determine availability of arming. But I think fail with message is likely better since you can give user friendly text back like "Cannot arm without GPS lock" or something like that.

SamChenzx, 2016-05-23:
@DonLakeFlyer  I favor that GQC allows users to arm motors without GPS lock. Besides, can you also check these two issues, [#3385](https://github.com/mavlink/qgroundcontrol/issues/3385) and [#3384](https://github.com/mavlink/qgroundcontrol/issues/3384) thanks!

DonLakeFlyer, 2016-05-29:
@LorenzMeier Are you ok with changing this back to the way it was before?

LorenzMeier, 2016-05-29:
In general allowing the user to arm the system without allowing him to take off (because no position estimate is present) seems like a terrible idea. We can clean up the check, but I would like to avoid having users arming the system before they are really able to fly.

DonLakeFlyer, 2016-06-02:
Discussion in #3460

closed 2017-03-04 by DonLakeFlyer