mavlink/qgroundcontrol #3407: Seems to be getting a lot Vehicle did not respond to parameter request

DonLakeFlyer, 2016-05-26:
I get it the most right after a firmware upgrade. Need to check if that one if just the px4io flash delay.

dzett, 2016-05-26:
Same here:

Same issue here. PX4 Master build as of 05/25 configured as Tailsitter VTOL, QGC windows master downloaded just now.

1. QGC started, PX4 without battery power, connecting via USB --> Heartbeat received, but "vehicle did not respons to request for parameters, retrying"
2. PX4 already powered via USB only, waiting until booted (blue blinking), then starting up QGC --> "WARNING: SYSTEM BOOT INCLOMPLETE. CHECK CONFIG." _but_ heartbeat and parameters accessible
3. PX4 powered via battery, usb plugged in after boot, then starting qgc --> works
4. PX4 powered via battery, starting qgc, then plugging in usb --> works
5. after 4.), removing usb and plugging it back in (leaving px4 powered via battery) also yields the parameter error message.

dogmaphobic, 2016-05-27:
When switching airframe type, after reseting the PX4, you get this error as well. Consistently.

DonLakeFlyer, 2016-05-27:
@dzett What OS are you running on?

DonLakeFlyer, 2016-05-27:
I can't repro this on Windows so this seems to be caused by the El Capitan hang causing something to go wrong as discussed in #3420

DonLakeFlyer, 2016-05-27:
And now I can't even repro on OSX!

DonLakeFlyer, 2016-05-27:
QGC is writing the correct bytes out to request the param list, but the firmware never seems to see it.

DonLakeFlyer, 2016-05-27:
Incoming bytes from Vehicle still workk correctly.

DonLakeFlyer, 2016-05-27:
@LorenzMeier I could use your help with this. When this happens can you see if the firwmare side stops receiving any bytes at all from QGC?

dzett, 2016-05-28:
@DonLakeFlyer I am seeing the above issues on Windows 10. I guess you could repro it at the end? I can confirm that for the cases where I see the issue with the params, the heartbeat send from the pix to qgc does show (status on orientation, compass,...).

DonLakeFlyer, 2016-06-02:
As far as I can tell this is related to #3225. Until that is resolved not worth digging more here.

dogmaphobic, 2016-06-06:
I am right now in a situation where I'm getting this 100% of the time and it is not related to the El Capitan hang issue. Note that each time I reboot the Pixhawk I get some different behavior. In the current state, I can make a standard (serial) connection, read some bytes and be done with it just fine (code below). However, if I launch QGC, I get this *Vehicle did not respond to request for parameters, retrying* error. Every time with no exception.

Note that the connection itself is working fine. There were no delays. I booted QGC and it immediately connected and started receiving data from the flight controller. Only the parameters are "hung". You can see that the MavLink inspector shows messages coming in just fine:

[image]

That does not mean the El Capitan hang issue is not real. I do see that a lot. But the behavior changes each time I reboot the flight controller. In this one instance, since the last reboot of the Pixhawk, I am getting this QGC behavior consistently 100% of the time while not having any (serial connection) issues connecting to the Pixhawk when using something else.

The code I'm using for testing the serial connection:

```
//
//  main.cpp
//  serialtest
//
//  Created by Gus Grubba on 5/29/16.
//

#include <stdio.h>
#include <fcntl.h>
#include <termios.h>
#include <unistd.h>
#include <sys/select.h>
#include <ctime>
#include <iostream>

#define SYSCALL(A) do { ret = A; if (ret == -1) { perror(#A); return -1; } else printf("%s returned %d\n", #A, ret); } while (0)
int ret;

int main()
{
    const clock_t begin_time = clock();
    struct termios tio;
    char buf[256];
    int fd = open("/dev/cu.usbmodem1", O_RDWR | O_NOCTTY | O_NONBLOCK);
    fd_set rfds, xfds;
    struct timeval to;
    to.tv_sec  = 5;
    to.tv_usec = 0;
    
    SYSCALL(tcgetattr(fd, &tio));
    
    cfmakeraw(&tio);
    
    tio.c_cflag = CS8|CREAD|CLOCAL;
    tio.c_cc[VMIN]  = 1;
    tio.c_cc[VTIME] = 1;
    
    cfsetispeed(&tio, B115200);
    cfsetospeed(&tio, B115200);
    
    SYSCALL(tcsetattr(fd, TCSANOW, &tio));
    
    FD_ZERO(&rfds);
    FD_ZERO(&xfds);
    FD_SET(fd, &rfds);
    FD_SET(fd, &xfds);
    
    int ret = select(fd + 1, &rfds, NULL, &xfds, &to);
    if (ret == -1)
        perror("select");
    else if (ret > 0)
    {
        if(FD_ISSET(fd, &rfds)) {
            std::cout << "Ready to read\n";
            size_t rd = read(fd, buf, 256);
            std::cout << "Read " << rd << " bytes\n";
        }
        if(FD_ISSET(fd, &xfds))
            std::cout << "Exception!\n";
    }
    else
        std::cout << "Timed out!\n";
    close(fd);
    std::cout << "Elapsed: ";
    std::cout << float(clock () - begin_time) /  CLOCKS_PER_SEC;
    std::cout << "\n";
    return 0;
}
```

dogmaphobic, 2016-06-06:
I should mention that there are hundreds of those messages as it keeps *retrying*:

[image]

dagar, 2016-06-06:
I'm getting this with only some hardware on Linux and Windows connected through USB. What firmware and version is everyone using?

dogmaphobic, 2016-06-06:
Mac OS, master Firmware, master QGC.

DonLakeFlyer, 2016-06-06:
Like I said way up above. I need to help from someone on the firmware side to debug whether the vehicle just stops getting bytes from QGC. QGC is defintely putting the bytes out on the serial port.

dogmaphobic, 2016-06-06:
Yes, it seems QGC keeps sending these requests but never gets any response. I can trace the calls all the way to serial:write() but it seems the flight controller is either not getting them or ignoring it. For instance, I setup a "mission" and tried uploading it. Same issue:

[image]

DonLakeFlyer, 2016-06-06:
As far as I can tell all communication from QGC to Vehicle just stops work on the Vehicle side of things.

dogmaphobic, 2016-06-06:
And just for completion sake, it's quite random. I just rebooted the Pixhawk and started QGC (after the Pixhawk was already in ready mode). No issues. Immediate connection, and all communications working. Proceeded to reboot yet again (after having closed QGC). Booted QGC, again after the Pixhawk came alive (after the bootloader mode) and this time, I got the "El Capitan Hang". It took for ever to connect, and when closing, it took for ever to disconnect. Once QGC was down I tried my code above. Same thing. Took forever to connect (and I found an error with my timer showing bogus elapsed time :P)

dagar, 2016-06-06:
Hard to say for sure because I don't have a consistent reproduce, but is it possible requesting AUTOPILOT_CAPABILITIES in the Vehicle constructor is interfering with the param sync?

https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/Vehicle.cc#L200

DonLakeFlyer, 2016-06-06:
I've debugged through this. The commands are making it out the serial port on the QGC side. As far as I can tell the Vehicle never receives them, or doesn't react to them for some reason. Hence I need someone to check if the firmware is receiving any bytes from QGC side of things when this happens.

dogmaphobic, 2016-06-06:
How can I set Telem2 to put out a shell instead?

LorenzMeier, 2016-06-06:
You don't. You need to connect Serial 4/5: http://dev.px4.io/advanced-system-console.html

dogmaphobic, 2016-06-06:
I know about Serial 4/5. But that would require building a cable and taking the drone apart to reach the Pixhawk. I have a nice, ready made patch cable connected to Telem2 already. I was just wondering if there was a way to set it up as a shell tty. Oh well, nice try though :P

NaterGator, 2016-06-06:
@dogmaphobic you're looking for this:
https://github.com/PX4/Firmware/blob/f9df60919e4722edade53b066d25248af9d73fd6/nuttx-configs/px4fmu-v2/nsh/defconfig#L544

dogmaphobic, 2016-06-06:
> you're looking for this:

I figured it had to be there some where :)

Thanks for pointing it out.

NaterGator, 2016-06-12:
Fwiw I was recently having this happen while running through mavproxy. I tried running param fetch in mavproxy console and QGC sprang to life. I don't know if it is directly relatable, but that speaks to an issue still present in QGC to me.

DonLakeFlyer, 2016-07-01:
@dzett Is this still happening for you with latest master on both firmware and QGC?

dzett, 2016-07-03:
@DonLakeFlyer, tried all scenarios that I had described originally with today's master. Works fine for all scenarios (case 5 got stuck once, but couldn't repro consistently). Good from my perspective, thanks!

DonLakeFlyer, 2016-07-07:
@NaterGator Any news on repro for this? I'm not seeing it any more in general use.

closed 2016-11-11 by DonLakeFlyer