mavlink/qgroundcontrol #3411: Multiple UAVs Constantly Disconnecting

mattforeman, 2016-05-26:
I have two Pixhawk driven UAVs that I want to connect simultaneously to Qgroundcontrol.

If I connect them over the 915Mhz telemetry individually, the connect fine. As soon as I connect the second one, it gives the error: " Vehicle did not respond to request for parameters, retrying". From there it goes through various states of connecting and disconnecting.

I have attached a video and a log in hopes that it might help.

youtu.be/ziuhPR2oaEA

[LinkManager_MultiVehicleManager_Log.txt](https://github.com/mavlink/qgroundcontrol/files/285191/LinkManager_MultiVehicleManager_Log.txt)

dogmaphobic, 2016-05-26:
2 SiK transmitters and 1 receiver? I didn't know that it would work that way. Assuming it does work, wouldn't the (double) data rate saturate its bandwidth?

DonLakeFlyer, 2016-05-26:
Good point. Did you try over USB?

mattforeman, 2016-05-27:
After assuming incorrectly that the telemetry modules work more like a bluetooth setup where you have one host and can connect multiple peripherals, I figured out the problem.

I have 3 sets of YKS telemetry modules from Amazon that all shipped with a NetId of 25, so they were all trying to connect to each other. I configured them into 3 pairs of unique NetIds and now everything is working.

closed 2016-05-27 by mattforeman