mavlink/qgroundcontrol #3413: Airspeed sensor calibration button has gone missing (master May 22 still had it)

dzett, 2016-05-26:
Hi there,

Since I applied a master of qgc (windows) yesterday May 25 (also checked against a build just downloaded just 10 mins ago), the airspeed sensor calibration button has gone missing (pixracer px4 master from may 22 configured as tailsitter vtol):

[image]

The Airspeed sensor is still shown on the summary and also is functioning in principle as can be seen:

[image]

I have reinstalled a master build downloaded on May 22 and that one still has the button and also allows calibration to take place.

(I had first accidentially set the circuit breaker for airspeed sensor but I have removed it since with no effect on the issue).

Thanks for looking into this,
Daniel

DonLakeFlyer, 2016-05-26:
This is because it's VTOL which QGC can get confused about.

@LorenzMeier Not sure how to treat VTOL. Should it always be both FW and MR from a UI settings standpoint. As another example, the new Safety screen forks the UI based on FW/MR. What should it do with VTOL?

DonLakeFlyer, 2016-05-26:
Need to review all Vehicle fixedWing/multiRotor code forks for correctness with respect to VTOL.

DonLakeFlyer, 2016-05-27:
This contributes to the problem as well after the QGC fix: https://github.com/PX4/Firmware/issues/4660

closed 2016-05-27 by DonLakeFlyer