mavlink/qgroundcontrol #3424: libdirectfb-1.2.so.9 is missing in AppImage

probonopd, 2016-05-27:
When trying the AppImage on Ubuntu 16.04, then I get

```
/tmp/.mount_r5K53X/qgroundcontrol: error while loading shared libraries: libdirectfb-1.2.so.9: cannot open shared object file: No such file or directory
```
and it does not launch. Possibly this could be remedied by copying libdirectfb-1.2.so.9 into usr/lib inside the AppImage at AppImage creation time.

dagar, 2016-06-12:
This was auto closed by the commit. Can you test it @probonopd? I'm not on 16.04 yet.

probonopd, 2016-06-12:
Happy to test, where can I download the build @dagar?

dagar, 2016-06-12:
https://s3-us-west-2.amazonaws.com/qgroundcontrol/builds/builds/QGroundControl.AppImage

probonopd, 2016-06-12:
On Ubuntu 16.04 LTS:
```
me@host:~$ Downloads/QGroundControl.AppImage 
/tmp/.mount_ZHI2uw/qgroundcontrol: error while loading shared libraries: libfusion-1.2.so.9: cannot open shared object file: No such file or directory
```

dagar, 2016-06-12:
@probonopd how about this one?
https://s3-us-west-2.amazonaws.com/qgroundcontrol/appimage/QGroundControl.AppImage

probonopd, 2016-06-12:
```
/tmp/.mount_K1E9qT/qgroundcontrol: error while loading shared libraries: libts-0.0.so.0: cannot open shared object file: No such file or directory
```

Did you know that you can test easily with [testappimage](https://github.com/probonopd/AppImageKit/blob/master/AppImageAssistant.AppDir/testappimage)?

Like this: `sudo testappimage some-ubuntu-live.iso Your.AppImage`

dagar, 2016-06-12:
Thanks @probonopd. You've got some cool scripts.

The fixed version will be on s3 shortly.
https://s3-us-west-2.amazonaws.com/qgroundcontrol/master/QGroundControl.AppImage

closed 2016-06-12 by dagar

probonopd, 2016-06-12:
Excellent, runs fine now.

commit 73844645 (2016-06-12): AppImage include libdirectfb-1.2.so.9 and cleanup

-fixes #3424
commit dbaf7d0d (2016-06-12): WIP appimage add libdirectfb-1.2-9

-fixes #3424
PR #3531 (2016-06-12): appimage add libdirectfb-1.2-9
-fixes #3424
PR #3538 (2016-06-12): AppImage add more libfusion, libdirect
-fixes #3424 (again)