mavlink/qgroundcontrol #3440: Something wrong with COM_LOW_BAT_ACT

DonLakeFlyer, 2016-05-29:
Shows up in Safety Summary panel as well. Not sure if it's meta data or what.

[image]

DonLakeFlyer, 2016-05-29:
This was from a new flash.

dogmaphobic, 2016-05-29:
Weird... That's a `FactComboBox` using `COM_LOW_BAT_ACT` as a source.

dogmaphobic, 2016-05-29:
I can't reproduce this. Master QGC and master PX4:

[image]

See what your `PX4ParameterFactMetaData.xml` looks like.

dogmaphobic, 2016-05-30:
Same thing in with the safety panel. @DonLakeFlyer, can you check your `PX4ParameterFactMetaData.xml` file?

[image]

DonLakeFlyer, 2016-06-14:
All working now. Must have been some bad meta data that went through in the past.

closed 2016-06-14 by DonLakeFlyer