mavlink/qgroundcontrol #3446: Send to vehicle causes connection lost sometimes

glody, 2016-05-30:
Test Environment
QGroundControl: Daily Build: v2.9.7b-46-g4af5756
Windows: Windows 8.1 64 Bit Laptop
Linux: Ubuntu 14.04 64 Bit Laptop
OS X: OS X 10.11.2 MacBook Pro
Android: Android 5.1.1 Nexus 9 Tablet
Purpose:
Check if the mission is able to send to vehicle 
Step:
1. Lanuch QGC
2. Connect to SITL
3. Set waypoints in mission page
4. Click "Send To Vehicle" 
Excepted result:
The mission is able to send to vehicle without error
Actual result:
Connection lost
[image]

DonLakeFlyer, 2016-06-02:
My guess is at this point SITL has crashed. When this happens can you shut down QGC. Restart it and see if it connects. If it doesn't then SITL has crashed or hung.

glody, 2016-06-12:
Yes, you are right. Restart QGC unable reconnect to SITL. It should be SITL issue. thanks.

closed 2016-06-12 by DonLakeFlyer