mavlink/qgroundcontrol #3447: Waypoints layer place front of buttons layer in mission page

glody, 2016-05-30:
Test Environment
QGroundControl: Daily Build: v2.9.7b-46-g4af5756
Windows: Windows 8.1 64 Bit Laptop
Linux: Ubuntu 14.04 64 Bit Laptop
OS X: OS X 10.11.2 MacBook Pro
Android: Android 5.1.1 Nexus 9 Tablet
Purpose:
Check if the mission page shows normally
Step:
1. Lanuch QGC
2. Connect to SITL
3. Set waypoints in mission page
4. Check the position about waypoints and buttons
Excepted result:
Waypoints layer should place behind the buttons layer 
Actual result:
Waypoints layer place front the buttons layer
[image]

DonLakeFlyer, 2016-06-01:
Not sure what daily build that is. I fixed this a couple days ago. Can you double check. Thanks.

glody, 2016-06-03:
Checked on QGC Version: v2.9.7b-125-gce7035b, this issue unable to reproduce. Close it.

closed 2016-06-03 by glody