mavlink/qgroundcontrol #3453: Changing wifi channel

andrejpan, 2016-05-31:
I am using QCG 2.9.7b. At wifi settings menu I can not change wifi channel.
[image]
At parameter menu I can change it with no problem
[image]

dogmaphobic, 2016-05-31:
You must be running an old version of the ESP8266 Firmware. The UI prevents changing the channel if you have the unit setup in *Station Mode*. With that said, the UI should detect that and not just block you if you are running an old version.

The current version is 1.0.11 (you can check it on the summary page)

dogmaphobic, 2016-05-31:
Done!

closed 2016-05-31 by dogmaphobic