mavlink/qgroundcontrol #3456: QGC: Mission Edit problems.

tubeme, 2016-06-01:
Hello everybody,

Just to report some problems when programming large missions.

In the bottom bar, where I get info on the alt and distance of waypoint, there is no way I could see after point 58. Probably this number depends on the resolution. I'm fullHD so 58 points :)

Also when the waypoints on the right, fill the height of the window, each new waypoint is hidden, under the window so you cannot see rely the heidht of the waypoint except when you move the mouse and slide the waypoints.

You can see:

[image]

Also when you make new waypoints it did not remember the height from the previous one if you made some corrections. When you make corrections to a waypoint you have to explicitly hit TAB in order to move the focus on another box. Then it remembers the previous waypoint and places its altitude as height value.

DonLakeFlyer, 2016-06-01:
Height map athe the bottom should be left/right scrollable with the mouse.

NaterGator, 2016-06-01:
@DonLakeFlyer I think maybe @tubeme is suggesting we scroll that flickable to the inserted item's position too.

DonLakeFlyer, 2016-06-01:
Ah, good point.

tubeme, 2016-06-01:
@DonLakeFlyer It is good when you have mouse, but on the field I don't use one... Also I want to see at this bar and on the right waypoints ribbon the current waypoint that I'm editing now. It was hell to set up this 78 point mission...

NaterGator, 2016-06-01:
@DonLakeFlyer is there a reason we can add onValueChanged handling [here](https://github.com/mavlink/qgroundcontrol/blob/c29de21aa53fb733ef995af8f0e944757934b2a5/src/FactSystem/FactControls/FactTextField.qml#L25)? Obviously we should use the validator in a non-intrusive way so that we don't set invalid values to the fact and don't prevent the user from continuing to edit the field to fix the value. 

That would address @tubeme's other outstanding concern.

DonLakeFlyer, 2016-06-01:
> It is good when you have mouse, but on the field I don't use one

How do you use QGC without a mouse? How are you selecting things? You can scroll with the a trackpad or touch as well.

DonLakeFlyer, 2016-06-01:
> Also when you make new waypoints it did not remember the height from the previous one if you made some corrections. When you make corrections to a waypoint height you have to explicitly hit TAB in order to move the focus on another box. Then it remembers the previous waypoint and places its altitude as height value.

Arae you saying that the waypoint height change is not being saved but gets lost? Or just that the use a previous altitude for new waypoints thing is not working. Or both?

DonLakeFlyer, 2016-06-01:
@NaterGator I don't quite understand what you are getting at? What do you want to do in onValueChanged?

NaterGator, 2016-06-01:
> @NaterGator I don't quite understand what you are getting at? What do you want to do in onValueChanged?

I was referencing this:

> Also when you make new waypoints it did not remember the height from the previous one if you made some corrections. When you make corrections to a waypoint height you have to explicitly hit TAB in [...]

What he's talking about is typing in a value in the altitude fact text field and then _not_ hitting enter before dropping another point. valueChanged signal fires continuously as data is entered, editingFinished is fired when focus is lost or enter is pressed.

DonLakeFlyer, 2016-06-01:
Ok I get it now. What I think needs to happen is that I need to figure out how to get focus to work correctly. When you click in the map to add a new waypoint or click a toolbutton the focus should leave the previous control but it didn't. If the focus left, it would also finish the edit.

tubeme, 2016-06-02:
> How do you use QGC without a mouse? How are you selecting things? You can scroll with the a trackpad or touch as well.

I'm using it in the most simple way. By clicking on the buttons, holding and then drag with the pad. I select things by the buttons of the pad. I don't use the scroll function of the pad, because some of my laptops have it some don't So i don't bother to install the drivers for the pad and thus no scrolling. 

If the bottom bar with the waypoint heights is scroll-able, there should be an indication for this, and probably some left and right arrows to move around if you don't use scroll function.

tubeme, 2016-06-02:
@DonLakeFlyer 

> Arae you saying that the waypoint height change is not being saved but gets lost? Or just that the use a previous altitude for new waypoints thing is not working. Or both?

It is hard to explain but I will try. 

I have a mission programmed. I'm on the Waypoint 20. Until now the points were programmed at 100 meters. from 20 I want to rase the altitude with 40 meters. So I create waypoint 20 and in the Altitude dialog I change the altitude to 140 meters, then after the edit the cursor is active and blinking in the altitude dialog. SO if i dont hit TAB after the edit to 140, in order to remove the cursor from the altitude dialog, and in this condition I create a new wyaipoint 21 the it will be with default altitude of 100 meters, like it did not remember my edin in the waypoint 20 but rather uses the preset altitude of point 19. Actually i tried to hit enter after i make the edit to 140 but it did not solve the problem. The only way around is hitting TAB and removing the focus on the altitude dialog.

Is it understandable now?

tubeme, 2016-06-02:
@NaterGator 

> What he's talking about is typing in a value in the altitude fact text field and then not hitting enter before dropping another point. valueChanged signal fires continuously as data is entered, editingFinished is fired when focus is lost or enter is pressed.

Enter does not work! Only TAB.

DonLakeFlyer, 2016-06-02:
> If the bottom bar with the waypoint heights is scroll-able, there should be an indication for this, and probably some left and right arrows to move around if you don't use scroll function.

Just click and drag it to scroll.

tubeme, 2016-06-02:
I don't really like the waypoint ribbon at right of the screen. It is very cumborsome to use. I loose my focus going back and forth. When I have to make large missions with lots of changes in the altitude etc, I have to move back and forth between selecting waypoints for edit and the boxes on the right. Why not hide all this boxes. And when you click on a waypoint, show me the box of the waypoint right at my mouse click right beside the waypoint. IE when you make waypoint active for edit just show me the altitude dialog at the map. Then hide it. There is no point in this:

[image]

What kind of information i get from this to help me get at one glance my flight plan??

[image]

It just takes place from my map. This is on FullHD res imagine if I have a smaller display...

tubeme, 2016-06-02:
This could be much more valuable to me as a pilot.

[image]

tubeme, 2016-06-02:
@DonLakeFlyer 

> Just click and drag it to scroll.

Thanks it works this way... Did not know it. Probably if it kept the focus to the last point and moved away the first points I could have tried to scroll it...

DonLakeFlyer, 2016-06-02:
> Probably if it kept the focus to the last point and moved away the first points I could have tried to scroll it...

Nate has the fix for that I hope.

Can you enter usability problems into a separate Issue? Let's keep this one just for bugs.

tubeme, 2016-06-02:
OK.

tubeme, 2016-06-02:
Sorry for my usability comments. I just have it as one experience on the airfield some bugs plus some usability problems...

DonLakeFlyer, 2016-06-02:
> Sorry for my usability comments.

Please keep them coming! Nothing gets better without people complaining. Just don't want them to get lost.

DonLakeFlyer, 2016-06-29:
The "not hitting enter" problem is going to have to push to 3.1. At this point I'm not quite sure how to fix it. But it's a larger issue that shows up in other parts of the ui as well.

DonLakeFlyer, 2017-04-13:
> The "not hitting enter" problem is going to have to push to 3.1.

FYI: This is fixed now.

DonLakeFlyer, 2017-04-13:
> If the bottom bar with the waypoint heights is scroll-able, there should be an indication for this, and probably some left and right arrows to move around if you don't use scroll function.

There are scroll indicators which show up no these and then go away. Also it auto-scrolls to the newly added items now.

DonLakeFlyer, 2017-04-13:
I think most of this is better now. Except for the right panel thing. In general I agree that the format is not that great. But not something we'll tackle now.

DonLakeFlyer, 2017-12-29:
Pulling this from above, since it's a great idea:
[image]