mavlink/qgroundcontrol #3460: Work out pre-arm messaging system

DonLakeFlyer, 2016-06-01:
Currently ArduPilot has a pre-arm messaging system which QGC knows about. PX4 stack does not. I'll fill in details in a bit.

DonLakeFlyer, 2016-06-01:
Ok, here is how ArduPilot works:
- ArduPilot sends out STATUS_TEXT messages on a continuous basis when the Vehicle is not ready to arm. The reasons for not being able to arm are many: sensors not calibrated, not enough satellites, compass variance, on and on. Each of these tends to have a circuit breaker to turn the check off. The status message is user friendly and tells the user what the problem is.
- ArduPilot will send the message once ever 10 seconds
- Solo build sends every second and uses these to drive the UI of their custom remote
- You don't have to attempt an Arm to start getting this messages. You start getting them as soon as you connect.
- All of the messages start with "PreArm:"
- QGC filters the "PreArm" messages in FirmwarePlugin placing the message in the center of the Fly map instead of using the normal ui mechanism.
- QGC times out the "PreArm" message on the Fly map if it doesn't get another "PreArm" message with that same text
- Note that this is all just visual indication. QGC does not know anything about what these messages mean and does not use them to prevent the user from attempting to Arm.
- If the user attempts to Arm the command will fail, and the vehicle will send command failure ack. With the reason for the failure then sitting as text in the middle of the map.

The key here is that QGC doesn't attempt to understand when a command, whether it be Arm or say Flight Mode change will fail. It's a slippery slope to try to make QGC too smart with respect to that sort of thing. I'd rather keep QGC dumb and use something like ArduPilot does or create a new thing. Whatever the new thing is it's needs a simple indication whether arm is available as well as a text message as to why (as opposed t0 having QGC check multiple things to make a determination).

JonReacher, 2017-10-16:
Any progress with this? It would be very helpful to have a ready to arm indication in QGC for PX4 for vehicles that have the autopilot/LED covered.