mavlink/qgroundcontrol #3461: Save PR Android APKs for testing via sideload

NaterGator, 2016-06-02:
I would be nice if we could get travis-ci to upload PR apks somewhere temporarily. Maybe another S3 folder similar to the PDB symbols generated for PRs that gets cleared automatically after N days? Doing so would make it a lot easier to test functionality implemented in PRs on Android devices (otherwise developers have to install a full android SDK and trust that their build + environment will match the CI system output).

dagar, 2016-06-02:
All builds from branches off the main repo are uploaded to s3. 

    https://s3-us-west-2.amazonaws.com/qgroundcontrol/BRANCH/qgroundcontrol.apk

Regular pull requests from another fork are more complicated because the travis-ci instance doesn't have access to the s3 credentials.

NaterGator, 2016-06-02:
I guess this is blocked by https://github.com/travis-ci/travis-ci/issues/5579

closed 2016-06-02 by NaterGator