mavlink/qgroundcontrol #3465: Param file github json download failed: % 1

glody, 2016-06-03:
Test Environment
QGroundControl: Daily Build: v2.9.7b-125-gce7035b
Windows: Windows 8.1 64 Bit Laptop
Purpose:
Check if the common parameters about the ariframe you select is able to load
Step:
1. Lanuch QGC
2. Connect to IRIS+
3. Select "X Style, Y6A Style: Quad, Hexa, Qcto, X8, Tri, Y6A" in Airframe item Setting page
4. Click "Load common parameters" and select "3DR QUAD X4 RTF"
Excepted result:
The common parameters is able to load
Actual result:
Param file github json download failed: % 1
[image]

NaterGator, 2016-06-03:
Is the issue you are reporting the broken argument in the error string or the fact that the JSON file download actually failed?

DonLakeFlyer, 2016-06-03:
Both problems are bugs

NaterGator, 2016-06-03:
There is another string bug [here](https://github.com/mavlink/qgroundcontrol/blob/0669604a14994bccde04f1dc96b4cdd870c0dd58/src/AutoPilotPlugins/APM/APMAirframeComponentController.cc#L247)

DonLakeFlyer, 2016-06-04:
I can't repro the download failure with latest daily. I re-wrote this code last week which should have fixed a bunch of things up.

DonLakeFlyer, 2016-06-12:
@glody Can you retry this with latest daily?

glody, 2016-06-13:
Retry on daily build below:
Windows/OS X:v2.9.7b-221-gd3df08b
Linux:v2.9.7b-220-g13bf81a
Android:v2.9.7b-233-g91fd6ae

All platforms work well except Windows. I created another issue to track windows issue #3550

DonLakeFlyer, 2016-06-14:
Should be fixed now

closed 2016-06-14 by DonLakeFlyer