mavlink/qgroundcontrol #3467: Delete "Default Tile Set" causes all sets lost

glody, 2016-06-03:
Test Environment
QGroundControl: Daily Build: v2.9.7b-125-gce7035b
Windows: Windows 8.1 64 Bit Laptop
Purpose:
Check if offline maps work correctly
Step:
1. Lanuch QGC
2. Add some new sets in Offline Maps item preferences page
3. Click "Default Tile Set" and delete it
Excepted result:
Only default tile set deleted 
Actual result:
All sets deleted

dogmaphobic, 2016-06-03:
What we need is a better label for this. *Default Tile Set* is too ambiguous. When you delete the "Default Tile Set", you are deleting every tile cached as it is the root of all tiles. The confirmation dialog tells you as much but it obviously is not good enough.

[image]

DonLakeFlyer, 2016-06-04:
I'd leave alone for 3.0 and figure out a rework for 3.1

DonLakeFlyer, 2016-12-01:
Text is now: "This will delete all tiles INCLUDING the tile sets you have created yourself.\n\nIs this really what you want?".

I think that's good enough.

closed 2016-12-01 by DonLakeFlyer