mavlink/qgroundcontrol #3469: Random crashes on exit within the Joystick code:

dogmaphobic, 2016-06-03:
Since my last merge, I've been getting random crashes on exit. It's around the Joystick code. I haven't had the time to track it down:

```
Process:               qgroundcontrol [94061]
Path:                  /Users/USER/*/qgroundcontrol.app/Contents/MacOS/qgroundcontrol
Identifier:            org.qgroundcontrol.qgroundcontrol
Version:               2.9.7b (129)
Code Type:             X86-64 (Native)
Parent Process:        Qt Creator [92414]
Responsible:           qgroundcontrol [94061]
User ID:               501

Date/Time:             2016-06-03 12:28:55.825 -0400
OS Version:            Mac OS X 10.11.5 (15F34)
Report Version:        11
Anonymous UUID:        8E8F0B38-7EB4-ECE1-32C4-8D02D21ACD98

Sleep/Wake UUID:       F137B1FA-344B-45CF-A6C5-004BF92D99D3

Time Awake Since Boot: 680000 seconds
Time Since Wake:       350000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called
*** error for object 0x7fa58f4657d8: pointer being freed was not allocated
 

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff9b591f06 __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff955864ec pthread_kill + 90
2   libsystem_c.dylib             	0x00007fff9b54b6e7 abort + 129
3   libsystem_malloc.dylib        	0x00007fff8dc2a041 free + 425
4   org.qgroundcontrol.qgroundcontrol	0x000000010db62e79 Joystick::~Joystick() + 169 (Joystick.cc:70)
5   org.qgroundcontrol.qgroundcontrol	0x000000010dc599b5 JoystickSDL::~JoystickSDL() + 21 (JoystickSDL.h:15)
6   org.qgroundcontrol.qgroundcontrol	0x000000010dc59965 JoystickSDL::~JoystickSDL() + 21 (JoystickSDL.h:15)
7   org.qgroundcontrol.qgroundcontrol	0x000000010dc59989 JoystickSDL::~JoystickSDL() + 25 (JoystickSDL.h:15)
8   org.qgroundcontrol.qgroundcontrol	0x000000010db673b8 JoystickManager::~JoystickManager() + 392 (JoystickManager.cc:45)
9   org.qgroundcontrol.qgroundcontrol	0x000000010db674d5 JoystickManager::~JoystickManager() + 21 (JoystickManager.cc:47)
10  org.qgroundcontrol.qgroundcontrol	0x000000010db674f9 JoystickManager::~JoystickManager() + 25 (JoystickManager.cc:40)
11  org.qgroundcontrol.qgroundcontrol	0x000000010dbdb1f7 QGCToolbox::~QGCToolbox() + 263 (QGCToolbox.cc:102)
12  org.qgroundcontrol.qgroundcontrol	0x000000010dbdb345 QGCToolbox::~QGCToolbox() + 21 (QGCToolbox.cc:110)
13  org.qgroundcontrol.qgroundcontrol	0x000000010dbaa769 QGCApplication::~QGCApplication() + 217 (QGCApplication.cc:343)
14  org.qgroundcontrol.qgroundcontrol	0x000000010dbaa7d5 QGCApplication::~QGCApplication() + 21 (QGCApplication.cc:344)
15  org.qgroundcontrol.qgroundcontrol	0x000000010dbaa7f9 QGCApplication::~QGCApplication() + 25 (QGCApplication.cc:335)
16  org.qgroundcontrol.qgroundcontrol	0x000000010db75887 main + 1847 (main.cc:258)
17  org.qgroundcontrol.qgroundcontrol	0x000000010da4f1e4 start + 52
```

dogmaphobic, 2016-06-05:
The crashes are consistent. It crashes on exit every time. A bit more info:

[image]

```
Releasing joystick: "SpaceNavigator"
qgroundcontrol(34040,0x7fff7ca3e000) malloc: *** error for object 0x105417c88: pointer being freed was not allocated
*** set a breakpoint in malloc_error_break to debug
```

dogmaphobic, 2016-06-05:
Stack at the crash point:

[image]

DonLakeFlyer, 2016-06-11:
I don't have a repro for this. Visual code inspection doesn't show anything wrong that I can see.

DonLakeFlyer, 2016-06-14:
@dogmaphobic Assigning over to you since you have a repro.

dogmaphobic, 2016-06-23:
I have a 3D mouse (3DConnexion SpaceNavigator). It shows up as a joystick in QGC. If I unplug it, it doesn't crash. With it plugged in, the crash is 100% reproducible but I have no idea what all that joystick code is doing. This is new (as of the time I entered this issue). I've had this mouse for years and never had an issue before.

dogmaphobic, 2016-07-01:
Done.

closed 2016-07-01 by dogmaphobic