mavlink/qgroundcontrol #3482: Flight widget should limit to one item/row on tiny screens.

dogmaphobic, 2016-06-05:
iPhone 4s

Allowing more than one row simply doesn't fit. The scrolling (left/right) feature is not working either. Adding another issue for that.

[image]

dogmaphobic, 2016-06-05:
Alright, I assumed the other "screen" (scroll right) was to show more items. I didn't realize it's something else. It's just that on mobile (even iPad) it's broken. What is this? Vibe?

[image]

dogmaphobic, 2016-06-05:
I guess we need to rethink this widget a bit :) Even on Desktop it can be made to look quite weird. There is no limit to what you can add and it clips it. It should either limit it based on the height or allow for distribution on separate "screens".

[image]

DonLakeFlyer, 2016-06-05:
The widget should allow you to scroll left/right to get to the different panels. Up/Down if the panel is taller than the available space.

dogmaphobic, 2016-06-06:
Up/Down doesn't seem to be working.
What is supposed to be in the "different panels"? Now all I see is "Not Available". I selected all the options in "Values Widget Setup".
Also, if you uncheck the "Large" checkbox, it also unchecks item itself (removing it from the list).

DonLakeFlyer, 2016-06-06:
First panel are values. Second panel is Vibe (vibration) widget. Vibe is ArduPilot only at this point. Waiting on vibration message to show up on PX4 side.

dogmaphobic, 2016-06-07:
For what's worth, the scroll up/down is not working for me. If I have more items than can be shown, it simply gets truncated and no scrolling is possible. I first assumed it would distribute itself into several panels (the left/right scroll).

HTRamsey, 2024-06-03:
Don't believe this is relevant anymore with that widget redesigned

closed 2024-06-03 by HTRamsey