mavlink/qgroundcontrol #3483: What do we do with tiny screens?

dogmaphobic, 2016-06-05:
As is, all but the Fly View is pretty much not usable on very small screens. At least on iOS, there is no way to limit beyond iPhone/iPad. If you say you support iPhone, you are required to support everything starting with the iPhone 4s.

Plan View (with home set some 9,367km away):

[image]

Radio setup:

[image]

Some screens, especially under settings allow flicking left and right but it makes for a very awkward user experience.

dogmaphobic, 2016-06-05:
iPhone 6 everything is quite fine:

[image]

[image]

DonLakeFlyer, 2016-06-05:
From the 4S emulator:

[image]

[image]

[image]
Can't to be able to scroll horizontally

[image]

[image]

dogmaphobic, 2016-06-05:
Screen density for the 4s is very different (in addition to the narrower width). If you switch the base font size to 9 points (instead of the default 11), things fit fine. I don't have an actual iPhone 4s to test how usable that is though:

[image]

[image]

DonLakeFlyer, 2016-06-06:
All I can say is that on a real iPhone 5 the default font is pretty small. Slightly smaller than the standard system font.

dogmaphobic, 2016-06-07:
Should we close this now?

DonLakeFlyer, 2016-06-07:
I'll close it when I'm done running through everything. PX4 stack is almost done. Now I need to go through ArduPilot setup.

closed 2016-06-12 by DonLakeFlyer