mavlink/qgroundcontrol #3486: USB Connection not working on Android

jzalger, 2016-06-06:
Connecting to a 3DR radio using USB OTG does not seem to work.
No messages appear requesting permission for the app to use the device, and under the communication links settings page, no USB or Serial devices are found.

Android 5.0.1
QGC Version: 2.9.7b-151-ge594d7d

dogmaphobic, 2016-06-06:
You probably have the 3DR radio set to always use some other app, in which case, it will never be available for QGC. When you first connect the 3DR radio, Android will ask you what to do with it. If you select an app and tell it to always use that, only that app will be able to see and use the USB device.

athertop, 2017-04-17:
This issue still appears to be relevant. On connecting 3dr radio modem with otg cable (Android 7,nexus 6p), and selecting qgrondcontrol when prompted to 'choose an app for USB device' qgrondcontrol loads fine but then  trying to add a serial device shows 'no ports available'.

booo, 2021-10-30:
Works for me on Android >=8. Please reopen if you still have a problem.

closed 2021-10-30 by booo