mavlink/qgroundcontrol #3506: Internal accelerometer is detected as joystick on Laptop

CarlOlsson, 2016-06-08:
We have had very random problems lately. If I connect to the airplane using my laptop the servos and motors constantly spikes and moves in arbitrary directions. This does not happen if we use another laptop. It seems that my computer has a built in accelerometer that is detected as a joystick and that causes random set points to be transmitted to the airplane over the telemetry link (thanks @tumbili for helping tracking down the problem). Is there a preferred way of disabling this at the moment?
[image]
The problem shows up on different version of QGC but right now I am on the daily build 9de744502d6711e69f955c8a181fe8a8

CarlOlsson, 2016-06-08:
In order to disable the Joystick in qgc I need to calibrate it, but I can't do that since it is only a "three axis" joystick. So now I actually don't find any way of disabling it.

CarlOlsson, 2016-06-08:
If I use QGC v2.7.1 it works

DonLakeFlyer, 2016-06-08:
Workaround: Clear your settings using the --clear-settings command line option.

closed 2016-06-11 by DonLakeFlyer