mavlink/qgroundcontrol #3508: 2.9.7b crashes on 14.04

pavel-kirienko, 2016-06-08:
QGroundControl 2.9.7b (and possibly earlier versions as well) crashes immediately on Kubuntu 14.04 AMD64. Here's the console log:

```
$ ./qgroundcontrol-start.sh 
Logging ini file directory "/home/pavel/.config/QtProject"
Filter rules "qt.network.ssl.warning=false\n"
Settings location "/home/pavel/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Map Cache in: "/home/pavel/.cache/QGCMapCache100" / "qgcMapCache.db"
qml: setBasePointSize 4.609375
Orphaned log file count 0
Segmentation fault (core dumped)
```

Directory structure:

```
$ ll -h
total 8.7M
drwxrwxr-x  4 pavel pavel 4.0K Jun  8 16:57 ./
drwxrwxr-x 13 pavel pavel 4.0K Jun  8 16:51 ../
drwxrwxr-x  4 pavel pavel 4.0K May 11 02:30 flightgear/
-rwxrwxr-x  1 pavel pavel 8.7M May 11 02:30 qgroundcontrol*
-rwxrwxr-x  1 pavel pavel  259 May 11 02:30 qgroundcontrol-start.sh*
drwxrwxr-x  5 pavel pavel 4.0K May 11 02:30 Qt/
```

I tried to clear the map cache directory, didn't help. The application has not been run under any other user account.

The core dump is attached. [core.qgroundcontrol.24995.tar.gz](https://github.com/mavlink/qgroundcontrol/files/304915/core.qgroundcontrol.24995.tar.gz)

DonLakeFlyer, 2016-06-08:
Can you try a a daily build?

pavel-kirienko, 2016-06-08:
Yes, latest daily from qgroundcontrol.org failed identically.

dagar, 2016-06-08:
The daily and 2.9.7b are both working fine on my Ubuntu 14.04 machine. We don't have symbols for the linux builds. Could you attach the backtrace?

DonLakeFlyer, 2016-06-08:
@dagar Can you work your way through this one? You know me and Linux.

dagar, 2016-06-08:
Sure. I'm not feeling super motivated to debug a core without symbols, but if I can get more info from @pavel-kirienko or a local reproduce I'm sure we can figure it out.

DonLakeFlyer, 2016-06-08:
Maybe turn on all logging and see how far it gets

dagar, 2016-06-08:
@pavel-kirienko can you try `./qgroundcontrol-start.sh --logging:full`?

pavel-kirienko, 2016-06-14:
I noticed that a bunch of other applications dependent on Qt were also failing similarly. Turned out there was a problem with my OS, unrelated to QGC.

Sorry for the confusion.

closed 2016-06-14 by pavel-kirienko