mavlink/qgroundcontrol #3509: iOS TestFlight users

DonLakeFlyer, 2016-06-08:
Keeping the list here while we are getting things set up:
david@codenoevil.com

mhkabir, 2016-06-10:
kabir@uasys.io

dkemxr, 2016-06-10:
Dkemxr@yahoo.com

billbonney, 2016-06-10:
@billbonney email in my profile page

Texzin, 2016-06-10:
texzin@gmail.com

DonLakeFlyer, 2016-06-11:
Letting folks know that this email address needs to be your Apple Store registered email address.

richerlariviere, 2016-06-11:
richerlariviere@gmail.com

jcooper21, 2016-06-11:
Justincooper9@gmail.com

DonLakeFlyer, 2016-06-16:
Users are added. First build has been submitted for review. We'll see how long it takes to get out of review.

jcooper21, 2016-06-16:
What's the best way to connect 915mhz radio to an iPad?

Texzin, 2016-06-16:
I think wifi is the only connection for now. I dont think the ipad supports
the usb that is needed to connect the radio

jcooper21, 2016-06-16:
Thanks. Bluetooth should work as well though?

Texzin, 2016-06-16:
I think I have a bluethooth module laying around somewhere. Gona have to
test that.

Texzin, 2016-06-16:
I have a ipad mini 4 that am using. QGC looks great on it.

jcooper21, 2016-06-16:
To connect to a 915 radio we should just need a lighting connector to mini USB correct?  DJI connects using lighting as well.

Texzin, 2016-06-16:
I dont think ios supports usb otg..

jcooper21, 2016-06-16:
I tested it just now using a female to female USB adapter connecting telemetry to the lighting cable and it doesn't power on the telemetry. The lighting port is designed for accessories. @donlakeflyer are you able to provide any insight?

billbonney, 2016-06-17:
You cannot connect either Bluetooth Classic (ie. Bluetooth 2.0) or the use the Lighting Port without a approved MFI device. i.e. Made for iDevices. Your only option is using WiFi or Bluetooth Low Energy (BLE aka Bluetooth 4.0)
QGC docs are here http://qgroundcontrol.io not sure where the info is about WiFi connectivity though.

billbonney, 2016-06-17:
@Texzin @jcooper21 I've used this is the past to connect a telemetry radio. http://redpark.com/redpark-ttl-serial-cable-simplifies-connections-to-arduino/

Texzin, 2016-06-17:
Just a thought if you are looking for range...might be able to set up a a wifi bridge with the bridge using a rf radio to communicate to drone

jy723, 2016-06-17:
252969186@qq.com,

DonLakeFlyer, 2016-06-17:
Folks. Lets keep this issue clear for people signing up for testing. Any issue go to gitter: https://gitter.im/mavlink/qgroundcontrol or google group: qgroundcontrol@googlegroups.com.

DonLakeFlyer, 2016-06-17:
To answer the connection question: The only connection supported at this point is WiFi. The only other real option is bluetooth, but that isn't supported right now. Any sort of USB thing won't work per Bill's explanation (unless someone gets something approved by Apple).

CarlOlsson, 2016-06-17:
We would like to sign up for the iOS beta testing
aeroparts01@gmail.com
carlolsson.co@gmail.com

DonLakeFlyer, 2016-06-17:
Latest set of users added

fw8, 2016-06-18:
Please add me to the iOS beta list 

florian@roedling.de

____________________________________________________________________________

FreiNet Gesellschaft fuer Informationsdienste mbH
Loerracher Strasse 5 a, 79115 Freiburg.
Telefon 0761/496-1700, Fax 0761/496-1790
http://www.freinet.de

Registergericht AG Freiburg i. Br. - HRB 4758
Geschaeftsfuehrung: Manfred Neufang
Steuernummer 06425/40959

Die FreiNet GmbH in Freiburg ist ein Unternehmen der Firmengruppe „Badisches Pressehaus GmbH & Co KG“ und zuständig für den technischen Betrieb der Plattformen www.badische-zeitung.de und www.fudder.de.
____________________________________________________________________________

Diese E-Mail enthaelt vertrauliche und/oder rechtlich geschuetzte Informationen.
Wenn Sie nicht der richtige Adressat sind oder diese E-Mail irrtuemlich erhalten haben,
informieren Sie bitte sofort den Absender und vernichten Sie diese E-Mail.
Jegliche Art der Verwendung, Vervielfaeltigung oder Weitergabe ist nicht gestattet.

bsauce42, 2016-06-18:
Add me to iOS beta 
bsauce42@gmail.com

colinhelms, 2016-06-19:
i'd love to try the beta! colinhelms@mac.com

DavidIngraham, 2016-06-19:
I would also like to beta test! davingrahamd@gmail.com

winstonsaz, 2016-06-20:
Hi, i has emerged a question, it will be possible to use "iDroneLink" with iOS QGC? the apps for device is not receiving support from the developers, that device was approved by Apple.

[image]

billbonney, 2016-06-20:
To use that iDroneLink requires permission from the fighting walrus guys to get your app signed by apple. 

In practice you'll get better results for long range using a WiFi bridge.  Also this conversation should be had on the QGroundControl mailing list

mboland, 2016-06-20:
mike@uavs.com.au

armattanpilot, 2016-06-21:
Can you please sign me up for the beta ios qgroundcontrol?
thijs@centeni.nl

mdaniels5757, 2016-06-21:
I'll help out. Email is mdaniels5757@gmail.com.

mboland, 2016-06-21:
Received my activation code but when I went to use it, about 12 hors later, it said expired.
What is the time limit and can I get another activation code?

DonLakeFlyer, 2016-06-22:
What's you email. I'll delete you and re-add.

djrocketodd, 2016-06-22:
todd@goascend.ca would like to test both android and ios

mboland, 2016-06-22:
Thanks Don, it's mike@uavs.com.au

karanmartian, 2016-06-22:
Hi Don,
Can you please add my address too for testing the android and ios apps? Its karankamdar@gmail.com

DonLakeFlyer, 2016-06-22:
@mboland I deleted and re-added you

DonLakeFlyer, 2016-06-22:
@djrocketodd For android testing look here for details: http://www.qgroundcontrol.org/downloads

karanmartian, 2016-06-22:
Hi Don,

I received the link and installed the apps. Thanks, but I hit a snag with the redeem code. First when I was entering the redeem code it repeatedly said "TestFlight is currently unavailable. Try again later". And now its saying "Could not accept invite. Redeem code expired". Has anyone experienced this issue before. I really like what I see so far with the QGroundControl and would like to continue testing it past the 55 days expiry.

DonLakeFlyer, 2016-06-22:
Yes, @mboland was having the same problem. Let me give it a try.

DonLakeFlyer, 2016-06-22:
Hmm I can't try it myself since I"m the admin. @mboland Can you let us know if the delete/re-add worked?

djrocketodd, 2016-06-22:
My code was expired, no worries as I am still getting my airframe set up, I can revisit you once its all tuned up.

DonLakeFlyer, 2016-06-22:
Hmm, not sure what is going on that. Build says it still has 55 days left on it. We may need to push up a new build and see if that resets things.

mboland, 2016-06-22:
Just entered the new redeem code and it took it.
Hope to start looking at it today.

karanmartian, 2016-06-22:
great! Don, could u do the same for me as well? i guess the delete and add
method

DonLakeFlyer, 2016-06-22:
@karanmartian Can you give me your email again?

tops4u, 2016-06-22:
Interessted... Currently I only have a BLE Device which I can Connect to the Telemetry. Can QGC handle BLE Devices?

Douggieboy, 2016-06-22:
DPratt5546@aol.com.
I use Mac exclusively. I want to use iPad and iPhone for f12e Devo tx with Tali H500 and 3dr x8+ drones. Thanks

karanmartian, 2016-06-23:
Hi Don,

My email is: karankamdar@gmail.com

tkschuler, 2016-06-23:
tkschuler@yahoo.com

karanmartian, 2016-06-23:
Hi Don,
Is there a new redeem code. I got the invite but there is no redeem code
instructions like. Do i have to uninstall and install again?

DonLakeFlyer, 2016-06-23:
Don't know. I'm not that familiar with the details of Test Flight. We'll be pushing up a new build soon. That should trigger things again I believe.

CarlOlsson, 2016-06-23:
Hi Don,

Couls you please remove and add again:
aeroparts01@gmail.com
I never received the email on this address and looking forward to try it:)

DonLakeFlyer, 2016-06-23:
New build is working it's way to you. Probably show up tomorrow. Hopefully that will reset folks that are having problems.

glody, 2016-06-24:
Hi Don,

Please add me to the test list. My apple account is: glody.guo@gmail.com

karanmartian, 2016-06-24:
Thanks for the new push Don. The app still says 60 days to expire. I guess
I shud not worry about it since there will be new builds at least once a
month which will reset that counter?

Deathstroke5467, 2016-06-25:
marcello.j.barbieri@gmail.com 

Would be pleased to test out the app, cant wait to try it. Out of interest similar to the way you can use a joystick in APM to override and essentially use it instead of RC remote, will there be a feature to use iOS game pads to control the vehicles? Also, what about allowing the qgroundcontrol desktop app to relay to the iOS app, such that you could use a telemetry radio (not possible on iOS) and relay the commands through a computer such as the Pi running the qgroundcontrol? I apologize and would be interested to know if these features are already integrated.

Many Thanks,
Marcello

djrocketodd, 2016-06-25:
> marcello.j.barbieri@gmail.com <mailto:marcello.j.barbieri@gmail.com>

Hey I havent tried it yet but they are on order, Team Blacksheep crossfire. looks like you bluetooth from ios or android and it acts as the telemetry relay to pixhawk or apm  check it out and see what you think.

[154 more replies]

closed 2016-09-18 by DonLakeFlyer