mavlink/qgroundcontrol #3543: OSX package broken

dagar, 2016-06-12:
I'm not sure yet if this is build system weirdness, but the OSX travis-ci build is no longer creating a dmg.

Last good master build https://travis-ci.org/mavlink/qgroundcontrol/jobs/137009539 91fd6ae
Broken master build https://travis-ci.org/mavlink/qgroundcontrol/jobs/137011790#L2228 d3df08b

    Log: Deploying Qt frameworks found inside: ("qgroundcontrol.app/Contents/PlugIns/sqldrivers/libqsqlmysql.dylib")
    ERROR: no file at "/opt/local/lib/mysql55/mysql/libmysqlclient.18.dylib"
    Log: Created configuration file: "qgroundcontrol.app/Contents/Resources/qt.conf"
    Log: This file sets the plugin search path to "qgroundcontrol.app/Contents/PlugIns"
    hdiutil: create failed - Invalid argument
    make: *** [release/qgroundcontrol.app/Contents/MacOS/qgroundcontrol] Error 1

Diff between those builds. https://github.com/mavlink/qgroundcontrol/compare/d3df08b...91fd6ae

NaterGator, 2016-06-12:
https://bugreports.qt.io/browse/QTBUG-53738 ?

DonLakeFlyer, 2016-06-12:
I bet I broke with with my gstreamer changes maybe

DonLakeFlyer, 2016-06-13:
Fixed. Updates links as well.

closed 2016-06-13 by DonLakeFlyer