mavlink/qgroundcontrol #3548: Lost ability to Cache Maps

VeloSteve, 2016-06-12:
This refers to changes some time between the daily builds available on June 6th and today, June 12th.  This one is v2.9.7b-238-g9fgcb35, running on a laptop under Ubuntu 16.04.  After the initial problem I tried deleting anything which looked related to QGroundControl in .cache and .config and rebooting the machine.  No change.  I updated libXi which eliminated some error messages, but had no other effect.

Now when I navigate to the offline maps tab there is no visible map.  When I select "Add new set", the whole world is shown.  The option to show previews at the top corners is missing.  There are no visible zoom controls, and pinch-type screen touches don't work.   Oddly (at least to me) sliding two fingers up the mouse pad zooms in.  Note that pinch controls DO work in the initial view when the application starts.

The Min Zoom and Max Zoom sliders are locked in place at the left end of their ranges.  If I do anything on the control area where they are, the map resets to a very wide zoom (I see Africa on the left and Florida on the right).  The tile count is shown as one.

If I go ahead and click "Download", it creates a set which claims to have 12 tiles (not 1) and the size is shown as 4,194,610.2 terabytes.  I deleted it right away in order to not break the internet (just kidding).

With the June 6 build I could cache tiles, but there were issues with unavailable tiles.  Now I can't pre-cache anything.  Several people and ROVs will be 6000 miles from home and 120 miles from the nearest cell tower or network connection in about 2 weeks.  It would be really nice to have some maps cached in advance.

DonLakeFlyer, 2016-06-12:
Hmm, I'm not seeing any issue on OSX. But I am seeing some on iOS. Looking.

Depending on your OS, two finger slide up and down simulates the mouse-wheel which is zoom in/out on a map.

VeloSteve, 2016-06-12:
Thanks for the mouse tip - I'm more of a desktop user most of the time.

DonLakeFlyer, 2016-06-13:
This bug doesn't bode well for cross platform testing of the same code! 

Works fine on: OS X, iPhone
Doesn't work on: Ubuntu, iPad

VeloSteve, 2016-06-13:
I just tried Windows 10 on the same laptop where Ubuntu is installed (dual boot).  On balance, I'd say it's better than a week ago, but not perfect.

Good: 
I was able to create new set.
The problem with errors + loaded tiles not equaling the total seems fixed.
I got a set with no errors reported for an area about 30 miles by 60, zoom level 9 to 19, in a remote part of the Pacific.

Bad:
Still no previews (minor)
The sliders work, but don't show a number for zoom level (less minor)
After viewing a set there's no affordance for going back to Offline Maps directly.  I can get there by clicking General and then Offline Maps again.
After creating a new set it doesn't appear in the list at first.  I tried the General/Offline trick a couple of times.  Once I got two copies of each old set and one of the new one.  I don't think the data is bad, just the display.

It's great to see you guys on top of the bug reports so quickly.

DonLakeFlyer, 2016-06-13:
> Still no previews (minor)
> The sliders work, but don't show a number for zoom level (less minor)

Previews and zoom level changes will be coming back. Things are in flux with small screen changes. In the version you are using the full map will change to show the min/max zoom level while you are holding down the min/max slider buttons. This doens't work well though. So it will be changed.

> After viewing a set there's no affordance for going back to Offline Maps directly.

Click the X

> After creating a new set it doesn't appear in the list at first.

I was seeing that happen before the latest changes as well. No fix for that yet. If you go out of Offline and back it clears itself up.

> I got a set with no errors reported for an area about 30 miles by 60, zoom level 9 to 19, in a remote part of the Pacific.

How many tiles is that? Trying to come up with some reasonable max that is allowed.

DonLakeFlyer, 2016-06-13:
Also there's a bug where the X is tricky to click for some reason. That should go away soon.

VeloSteve, 2016-06-13:
The X works, and the zoom preview isn't so bad now that I know what's happening.  The only real problem is when it happens to fall on water, which looks about the same at different levels.

The set I mentioned  is 1,724 tiles, but I wrote the wrong size. The area is actually about 1 mile by 2.  It includes a tiny island and the area immediately around it.  I could easily imagine having a dozen sets like this.  My biggest set is 54,214 tiles with 26,715 downloaded and the rest failed.  This is the one which is 30 miles by 60.  This may be excessive - I just did it to have an overview of the area we're operating in.  The high failure rate is apparently because Google has zoom level 18 in a lot of the area, but far from all.

Just noticed another thing.  When I look at the 1,724-tile set it shows the area I selected.  The huge one just shows the world even though it was originally defined to a much smaller area.

DonLakeFlyer, 2016-06-13:
> The huge one just shows the world even though it was originally defined to a much smaller area.

Yup. That's another bug. The new stuff is a bit of a mess. Working through it. It's all just a few days old.

DonLakeFlyer, 2016-06-14:
All fixed now. Thanks for the bug report.

DonLakeFlyer, 2016-06-14:
These things still exists: #3563

closed 2016-06-14 by DonLakeFlyer