mavlink/qgroundcontrol #3549: Compass unable to calibrate

glody, 2016-06-13:
Test Environment
QGroundControl: Windows/OS X:v2.9.7b-221-gd3df08b
Linux:v2.9.7b-220-g13bf81a
Android:v2.9.7b-233-g91fd6ae
Windows: Windows 8.1 64 Bit Laptop
Linux: Ubuntu 14.04 64 Bit Laptop
OS X: OS X 10.11.2 MacBook Pro
Android: Android 5.1.1 Nexus 9 Tablet
Purpose:
Check if the compass is able to calibrate. 
Step:
1. Lanuch QGC
2. Connect to IRIS+
3. Calibrate Compass in sensors item, settings page
Excepted result:
The Compass is able to calibrate
Actual result:
Calibration diagram unable to show on Windows, OS X and Linux. On Android, the calibration diagram is able to show, but left/right/up/down status about vehicle unable to recognize by QGC
[image]

DonLakeFlyer, 2016-06-13:
I fixed this today. Should show up in a daily by tomorrow. Keep in mind that currently OS X daily builds are not posting: #3543

closed 2016-06-13 by DonLakeFlyer

glody, 2016-06-13:
Thanks for your quick response!

DonLakeFlyer, 2016-06-13:
Thanks for the awesome testing and bug reports!