mavlink/qgroundcontrol #3553: Only half characters shows on dashboard in flight page on Android

glody, 2016-06-13:
Test Environment
QGroundControl: Android:v2.9.7b-233-g91fd6ae
Android: Android 5.1.1 Nexus 9 Tablet
Purpose:
Check if the dashboard shows correctly
Step:
1. Launch QGC
2. Connect to IRIS+
3. Check the display on dashboard in flight page
Excepted result:
All characters are shown correctly and completely
Actual result:
Only half shows about bottom characters. please check the snapshot for details.
[image]

DonLakeFlyer, 2016-06-13:
@glody An additional piece of information is what firmware you are using. ArduPilot or PX4. Are you mainly testing on ArduPilot or both?

glody, 2016-06-24:
@DonLakeFlyer I mainly test ArduPilot. Just use PX4 as reference.

DonLakeFlyer, 2016-07-07:
@glody Have you modified the value list for the first page on the instrument panel? Could you show me what that looks like?

glody, 2016-07-26:
Sorry for slow response. Please check the snapshot below.
[image]
[image]

DonLakeFlyer, 2017-03-19:
This was reworked. I think it's fixed now

closed 2017-03-19 by DonLakeFlyer