mavlink/qgroundcontrol #3563: Offline Map bugs

DonLakeFlyer, 2016-06-14:
* If you click between two created tile sets the map may not zoom correctly.
* After you create a new tile set, sometimes the tile set list doesn't show correct. If you click away to another settings screen and come back it will usually reset itself. Otherwise restart QGC and everything will be ok. Download may have paused though.

DonLakeFlyer, 2016-06-29:
@dogmaphobic Do you have time to look at #2. Seems like a signalling or list update problem. I could repro prior to changes as well. If you don't have time, push all of this to 3.1.

#1 is a Qt bug I believe, but not quite sure. It just fails to do the right thing. Not as worried about that one.

DonLakeFlyer, 2016-07-07:
@dogmaphobic What do you want to do wih #2?

dogmaphobic, 2016-07-07:
I missed these. For some reason my mail rules didn't flag this issue.

None of this was happening when I created the first UI though. I have not played with maps in a while. I'll take a look.

DonLakeFlyer, 2016-07-07:
I'm pretty sure I went backwards to check the old code and was able to repro, just not as easily. Not 100% sure since it was a while ago.

dogmaphobic, 2016-07-08:
Hmm, I can't reproduce any of these. I've created 8 sets so far and no issues.

DonLakeFlyer, 2016-07-08:
Ok, thanks. Let me try.

DonLakeFlyer, 2016-07-08:
I was able to repro #2. But I think what I thought was the list not showing correctly was just a delay in it showing up in the list. I added a new set. Clicked download. Then it took about 3 seconds for the new set showed up in the list.

dogmaphobic, 2016-07-08:
Once you "create" a new set, a call is made to the tile set engine, which starts the process on a separate thread. The set is created within the database and the download thread starts. As these things happen in the background, the UI (QML) gets updated through the *models*. It takes a bit of time to create the set (< 1s) and some time to start receiving tiles (also < 1s). Once the set is created, it should show up in the list (with 0 tiles downloaded). As tiles begin to arrive, the download count (and remaining count) start updating. If it's a very small set, it could all show up at once with the set already completed.

DonLakeFlyer, 2016-07-08:
Ok, so #2 is just timing. I couldn't get #1 to repro. Closing this.

closed 2016-07-08 by DonLakeFlyer