mavlink/qgroundcontrol #3569: GPS Lock for two vehicles not working correctly

DonLakeFlyer, 2016-06-14:
Okay So I have two drones and they are connected to the same QGC. Drone one is called Vehivle #1 and and drone 2 is called Vehicle #2. What I did is I connected both of them to the same QGC and on my MAVLink inspector in widgets I can see the mavlink messages of Vehicle #1 and Vehicle #2 which means that the yare connected to QGC. Next is if the GPS data of both the Vehicles in communicated to the QGC via mavlink or not ? So I observed that inside my mavlink inspector I can see GPS data of both the vehicles. Problem I am having is I can onlu see gps lock of one vehicle like it shows 8 satellites. But when I go to the flight view of vehicle #2 on the same QGC I can not see any gps lock. It says no GPS lock even though the mavlink messages are being communicated successfully from the Vehicle to the QGC.

DonLakeFlyer, 2016-06-14:
@musclesinwood I moved your issue here so we don't lose it

DonLakeFlyer, 2016-06-14:
I tried to repro this and it all worked correctlyf or two vehicles. Can you double check you message traffic? if it all still looks good. Save a telemetry log me for the session and attach.

Removing milestone until I can get a repro.

musclesinwood, 2016-06-14:
Dear Don,
Thank You. Appreciate it. You meant check my Mavlink inspector ?

DonLakeFlyer, 2016-06-15:
Just using mavlink inspector you won't be able to tell if a 3d fix is coming across from both Vehicles. So that's not a valid suggestion. I'll need a telemetry log to go any further with determining what is going on.

musclesinwood, 2016-06-15:
Al right. So,you need the telemetry log of both of the Vehicles ( Vehicle 1 & Vehicle 2 )? And where can I find the log file ? Is it on the QGC ? Can you please point me to the right place where those log fiels are located ?

DonLakeFlyer, 2016-06-17:
QGC should prompt you to save a log file when you disconnect the last vehicle.

DonLakeFlyer, 2016-06-17:
If it doesn't turn on "Prompt to save Flight Log data after each flight" in settings

DonLakeFlyer, 2024-10-13:
Re-open of there are log files to reproduce

closed 2024-10-13 by DonLakeFlyer