mavlink/qgroundcontrol #3581: Migrate to Qt 5.7

NaterGator, 2016-06-16:
This also includes support for Qt Charts via GPL3... huzzah!

http://blog.qt.io/blog/2016/06/16/qt-5-7-released/

dagar, 2016-06-16:
I'm going to try a minimal docker image for Linux and Android builds.

pritamghanghas, 2016-06-21:
possibly only thing blocking this update is being fixed here. Hopefully this will make it to 5.7 release.

NaterGator, 2016-06-21:
5.7.0 is already out the door. :grimacing:  Based on Qt's previous releases 5.7.1 will be timed about 3 months from now.

pritamghanghas, 2016-06-21:
> 5.7.0 is already out the door. 😬 Based on Qt's previous releases 5.7.1

I wasn't aware that is unfortunate.

pritamghanghas, 2016-07-02:
This has been fixed and will be cherry picked for 5.7.1. If someone is working on charts and can't wait. Just pick this one https://codereview.qt-project.org/#/c/161978/ and build 5.7 locally and good to go.

DonLakeFlyer, 2016-07-02:
Good news. We can move to 5.7 before we toss analyze

DonLakeFlyer, 2016-10-18:
We are going to move to 5.6.2 instead since it's an LTS version.

DonLakeFlyer, 2017-05-26:
This is all done now

closed 2017-05-26 by DonLakeFlyer