mavlink/qgroundcontrol #3583: No bluetooth connection available

FPVflyer, 2016-06-16:
Hi guys,
I just installed the iOS version on my iphone, but can't connect to telemetry as I only use the TBS Crossfire as link.

Would be possible to add the bluetooth connections  to Comm Links?

Regards

DonLakeFlyer, 2016-06-16:
@dogmaphobic This is a dup I believe.

FPVflyer, 2016-06-20:
What do you mean with "dup"? On Android the Bluetooth comm links is available and work fine.

DonLakeFlyer, 2016-06-20:
@dogmaphobic Do you think you could ping the TBS folks to see if this is possible or not?

sirPerna, 2016-06-21:
Hi

Crossfire supports BLE so it should be possible. It's already active but doesn't use the correct UUID yet. You should already be able to see it and bind. I'm happy to help!

dogmaphobic, 2016-06-23:
> Crossfire supports BLE so it should be possible. It's already active but doesn't use the correct UUID yet. You should already be able to see it and bind. I'm happy to help!

Good. That's the only way QGC would be able to access it from iOS. I will take a look at implementing BLE on iOS.

DonLakeFlyer, 2016-06-24:
Not holding 3.0 for this

DonLakeFlyer, 2016-09-14:
http://discuss.ardupilot.org/t/bluetooth-connectivity-on-android/10917/9

DonLakeFlyer, 2016-10-03:
This issue is for both iOS and Android BT problems.

DonLakeFlyer, 2016-10-03:
@dogmaphobic So I'm not quite sure where to start here. I have a BT thingy connected to my Pixhawk. Works fine on OSX. I'm trying to test on Android. I don't get a serial port to connect to even though I'm paired. Is it androids job to expose the BT connection as a serial port or is that QGC code someplace to is turning a BT connection into a serial port.

sirPerna, 2016-10-04:
I've tested it with two android devices and crossfire bluetooth. It works fine under android 4.4.2 but I can't get a connection with android 6.0. 
Let me know how I can help here.

Patnetpatnet, 2016-11-12:
Unable to connect BT. QGC Android app.

dogmaphobic, 2016-11-14:
I spent the last couple of days on this and I don't have an answer. It works fine up to Android 5.x Something changed in Android 6.x (and 7.x) that causes the connection not to work. I changed the entire code to match this [example](http://doc.qt.io/qt-5/qtbluetooth-btchat-example.html) and still nothing works. Using the scanning/connection method shown in that [example](http://doc.qt.io/qt-5/qtbluetooth-btchat-example.html), it takes forever to scan the services (about 30 seconds). If I take the resulting `QBluetoothServiceInfo` and immediately use it to make a connection, I still get a "Service Not Found" error. The target address and service UUID (ffcacade-afde-cade-defa-cade00000000) are all correct. The connection just won't work.

I can't get the Crossfire paired with my Mac. It always fails so I can't run any tests on the desktop.

[image]

iOS (BLE) will be a whole different ball game. The existing code will scan and find the Crossfire properly but the comms are all wrong. BLE communication doesn't use sockets. You have to use a completely different interface that sends and receives 31 bytes at a time (or something to that effect).

Bluetooth is such a pain in the neck. Tower, which uses the standard Android/Java interfaces for Bluetooth connects just fine. It has to be something with Qt.

All this using the Crossfire with firmware 1.44.

hackra76, 2016-11-15:
Could this change be a problem with Android 6.x:
Access to Hardware Identifier
To provide users with greater data protection, starting in this release, Android removes programmatic access to the device’s local hardware identifier for apps using the Wi-Fi and Bluetooth APIs. The WifiInfo.getMacAddress() and the BluetoothAdapter.getAddress() methods now return a constant value of 02:00:00:00:00:00.

To access the hardware identifiers of nearby external devices via Bluetooth and Wi-Fi scans, your app must now have the ACCESS_FINE_LOCATION or ACCESS_COARSE_LOCATION permissions:

WifiManager.getScanResults()
BluetoothDevice.ACTION_FOUND
BluetoothLeScanner.startScan()
Note: When a device running Android 6.0 (API level 23) initiates a background Wi-Fi or Bluetooth scan, the operation is visible to external devices as originating from a randomized MAC address.

https://developer.android.com/about/versions/marshmallow/android-6.0-changes.html?hl=ru#behavior-hardware-id

dogmaphobic, 2016-11-15:
Good try but no difference. I tried the permissions above. I proceeded to add ALL permission requests I found in the Tower app. Same results. It's either something wrong with Qt's Bluetooth implementation or some (undocumented) configuration that I'm missing:

```
 D QGroundControl: /Users/gus/github/work/qgroundcontrol/src/comm/BluetoothLink.cc:181 (void BluetoothLink::serviceDiscovered(const QBluetoothServiceInfo&)): Connecting to: "Crossfire 2507" "{ffcacade-afde-cade-defa-cade00000000}"
 W BluetoothAdapter: getBluetoothService() called with no BluetoothManagerCallback
 W QGroundControl: (null):0 ((null)): qt.bluetooth.android: Cannot retrieve SDP UUIDs for "" ( "21:61:9D:1B:D2:56" )
 W QGroundControl: (null):0 ((null)): qt.bluetooth.android: Cannot retrieve SDP UUIDs for "house" ( "00:1E:52:EB:86:C4" )
 W bt_btif : bta_dm_acl_change info: 0x10
 D bt_btif_dm: remote version info [00:04:3e:95:5c:0e]: 0, 0, 0
 E BluetoothRemoteDevices: state12newState0
 W bt_sdp  : process_service_search_attr_rsp
 E bt_btif_sock_rfcomm: find_rfc_slot_by_id unable to find RFCOMM slot id: 4
 W System.err: java.io.IOException: read failed, socket might closed or timeout, read ret: -1
 W System.err:    at android.bluetooth.BluetoothSocket.readAll(BluetoothSocket.java:685)
 W System.err:    at android.bluetooth.BluetoothSocket.readInt(BluetoothSocket.java:697)
 W System.err:    at android.bluetooth.BluetoothSocket.connect(BluetoothSocket.java:374)
 W QGroundControl: (null):0 ((null)): qt.bluetooth.android: Falling back to workaround.
 W QGroundControl: (null):0 ((null)): qt.bluetooth.android: Using found rfcomm channel 0
 W QGroundControl: (null):0 ((null)): qt.bluetooth.android: Invoke Resulted with error.
 W System.err: java.lang.reflect.InvocationTargetException
 W System.err:    at java.lang.reflect.Method.invoke(Native Method)
 W System.err: Caused by: java.io.IOException: Invalid RFCOMM channel: 0
 W System.err:    at android.bluetooth.BluetoothSocket.<init>(BluetoothSocket.java:190)
 W System.err:    at android.bluetooth.BluetoothSocket.<init>(BluetoothSocket.java:166)
 W System.err:    at android.bluetooth.BluetoothDevice.createRfcommSocket(BluetoothDevice.java:1369)
 W System.err:    ... 1 more
 W QGroundControl: /Users/gus/github/work/qgroundcontrol/src/comm/BluetoothLink.cc:223 (void BluetoothLink::deviceError(QBluetoothSocket::SocketError)): Bluetooth error QBluetoothSocket::ServiceNotFoundError
 W QGroundControl: (null):0 ((null)): qt.bluetooth.android: Workaround failed
```

hackra76, 2016-11-15:
It looks like this issue with Qt might be the problem:
https://bugreports.qt.io/browse/QTBUG-55035

As quoted here: "You have to ask the user explicitly during runtime whether he wants to grant the needed permission (i.e. ACCESS_COARSE_LOCATION)."

dogmaphobic, 2016-11-15:
The scanning process is working. I do get a full scan with proper returned addresses and services (UUIDs). It's the connection that fails.

sirPerna, 2016-11-15:
@dogmaphobic For BLE do you see following UUIDs? 
SERVICE_UUID "180f"
CHAR_TX_UUID "2a19"
CHAR_RX_UUID "2a19"

This is usually a battery status UUID and the module hijacks it by default.
Is it possible to force android to use SPP over BLE also for version 6.0 and newer?

dogmaphobic, 2016-11-15:
I found the (BLE) UUIDs for the Crossfire. What we don't have is support for BLE communication at all. The current Bluetooth comm uses sockets, which assumes a "normal" BT link. Something new would have to be written to support BLE.

DonLakeFlyer, 2016-12-04:
Regular bluetooth fixed now. BLE still NYI.

closed 2017-02-18 by DonLakeFlyer