mavlink/qgroundcontrol #3586: Dependency 403

lutherling12, 2016-06-17:
Upon provisioning 'dev', an error is returned stating that Qt5.5.1-linux.tar.bz2 doesn't exist.

After looking in the Vagrant file, line 31 seems to perform wget to retrieve this dependency:
su - vagrant -c 'wget --continue -q %{deps_url}/%{qt_deps_tarball}'

Checking the variables in .vagrantconfig.yml, the two parts of the url are:
'deps_url': 'https://s3-us-west-2.amazonaws.com/qgroundcontrol/dependencies'
'qt_deps_tarball': '/vagrant/Qt5.5.1-linux.tar.bz2'

If I perform the wget on my host machine with the expanded url, it returns a 403 Forbidden error. Is anyone else having this issue? Is there anything that can be done about it?

dagar, 2016-06-17:
As far as I know none of the regular developers or build system use vagrant so it's not surprising that it broke.
The correct URL is: https://s3-us-west-2.amazonaws.com/qgroundcontrol/dependencies/Qt5.5.1-linux.tar.bz2

Could you fix the vagrant config, verify it actually works, and submit a pull request?

lutherling12, 2016-06-17:
I don't mind, but I'm not sure where to find this Qt5.5.1-linux.tar.bz2 file that's bundled with qGroundControl. Is there anything special about it that's different from the official Qt5.5.1 release?

lutherling12, 2016-06-17:
I'll get right on it. Thank you.

dagar, 2016-06-17:
We've been hosting our own Qt because so far it's been quite a pain to automate the install. I'd like to stop doing it if possible.

lutherling12, 2016-06-17:
readme states that it **must** be Qt5.5.1. Is this because the official installer is a pain to automate and makes it hard to install the latest version? If not, and the dependency is strictly for that version, why not host a box that already has Qt5.5.1 on Atlas?

dagar, 2016-06-17:
There are a few issues. Typically we see little bugs between different versions of Qt and just don't have the resources to test and maintain compatibility with multiple versions.

For testing we use the free/open source tier of travis-ci which needs to install the build dependencies every single build.

What's Atlas?

lutherling12, 2016-06-17:
Atlas is the platform Hashicorp uses to host vagrant boxes.
atlas.hashicorp.com

DonLakeFlyer, 2016-06-24:
@dagar Is this fixed now?

dagar, 2016-06-24:
I haven't personally tested it, but #3598 should have fixed it.

closed 2016-06-24 by dagar