mavlink/qgroundcontrol #3588: Plane: Takeoff item on ArduPilot is out of whack

DonLakeFlyer, 2016-06-17:


DonLakeFlyer, 2016-06-17:
This issue is different. It ths actual positioning of the item. Not the abbreviation. It may have been a home position problem though. Still looking.

NaterGator, 2016-06-17:
@DonLakeFlyer the offline data for ardupilot is definitely incorrectly loading the data for MAV_CMD_NAV_TAKEOFF:

[image]

Note that _specifiesCoordinate is false.

NaterGator, 2016-06-17:
Here ya go: 
https://github.com/mavlink/qgroundcontrol/blame/fdb13156c0caa95a587775f9c60e4173e13b7a27/src/FirmwarePlugin/APM/MavCmdInfoCommon.json#L10

DonLakeFlyer, 2016-06-17:
ArduPilot doesn't support coordinate for takeoff. This was just a an issue with a bad planned home position which took the takeoff item along with it. Not really a bug. User error

closed 2016-06-17 by DonLakeFlyer

NaterGator, 2016-06-17:
The mission controller [does this](https://github.com/mavlink/qgroundcontrol/blob/fdb13156c0caa95a587775f9c60e4173e13b7a27/src/MissionManager/MissionController.cc#L171) automagically when planning the 2nd point in a mission. If it is desired behavior that is fine, just FYI.

DonLakeFlyer, 2016-06-17:
The assumption is that most missions start from the ground. Hence first thing you want is takeoff. If it's annoying people it can be removed.